Webreaper/SolisAgileManager #28: Extend cheaper session until prices go up

Webreaper, 2025-01-11:
If we have a cheapest session of: 

* 20p
* 21p
* 22p
* 21p
* 19p
* 20p

And then the next prices are the same or lower:

* 20p
* 19p
* 21p
* 22p
* 26p

Then include the first ones (20, 19, 21) into the cheapest slot, because we might as well use the grid during the cheapest period and maintain the battery charge level, and only stop charging when prices start to rise.

MartyMoose30053, 2025-01-25:
Is there a reason you chose to charge in a continuous slot?
When manually programming charging I would for example as in your first example (below)pause charging for the 22p then resume when it drops to 21p.

20p
21p
22p
21p
19p
20p

Having manually set these for nearly two years there is very often a spike in the price mid charge slot. The Solis only has 3 charge slots so not possible to avoid them unlike your App.

👍

Webreaper, 2025-01-25:
It already does that. It'll look for the cheapest 6 slots out of the 8 in the run up to the peak period.

I try to group charging slots in a single contiguous batch where possible, because it reduces the number of writes to the inverter. Too many writes could reduce the lifespan of the inverter (see "Technical Considerations" in the readme).

MartyMoose30053, 2025-01-25:
Ahh understood and makes perfect sense

Webreaper, 2025-01-25:
It's a tricky business!