Webreaper/SolisAgileManager #43: Raspberry PI specification required to run this

dqj999, 2025-01-16:
Can you clarify what is the required spec/model for a Raspberry PI to run this package?

Derek Johnson

Webreaper, 2025-01-16:
I know nothing about Raspberry Pis, I'm afraid. All I've done is build a package compiled for ARM64 Linux, that should run on a RPi. You may need to go and ask in an RPi forum to find out. 

The app doesn't use much memory/CPU/ etc though, so if I had to guess I'd say "any".

dqj999, 2025-01-16:
Thanks, according to this article, [here](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/) it looks like these PIs will run Arm64 packages:

Pi Zero 2
Pi 3
Pi 4

Webreaper, 2025-01-16:
I'll have a look into making an ARM64 Docker image - that might make deployment on RPi easier than manually copying binaries etc. https://github.com/Webreaper/SolisAgileManager/issues/44

closed 2025-01-16 by Webreaper

dqj999, 2025-02-15:
Some people have got docker images working successfully on Raspberry PIs.

Here's a link to one that is quite interesting:

[Docker on Raspbery PI project](https://github.com/eelmafia/octopus-minmax)

I don't claim to understand any of this but it might shed some light on why my attempts to make it work failed.

Webreaper, 2025-02-15:
Thanks, useful.