Webreaper/SolisAgileManager #46: Always charge below threshold price not working

Webreaper, 2025-01-16:
@dqj999 mentions that setting an "always charge below" value of 20p/kWh isn't resulting in slots cheaper than 20p being set to charge.

dqj999, 2025-01-16:
When these outputs were being created, the rates must have been estimated Agile prices since the official ones would not have been available till after 16:30 or later. I know that the estimated prices also are subject to being updated at several points in the day. Thus the charging schedule might need to change after being created. I know that this happens with the Ohme EV charging schedule which also uses Agile prices. Perhaps there is some inconsistency in the use of the Agile rates stored in the App which could possibly be out of date.

Webreaper, 2025-01-16:
The app pulls prices from Octopus every 30 mins, so if the prices were to change, they'd be updated. That said, I believe that Octopus guarantees the prices (for Agile anyway) once published.

The fact that this isn't working seems like a bug, I'll investigate a fix later.

dqj999, 2025-01-16:
> That said, I believe that Octopus guarantees the prices (for Agile anyway) once published.

Octopus don't publish Agile rates past 22:30 on the current day until 16:30 or later. There are websites that attempt to estimate the rates but these are usually incorrect by several pence in advance of the published rates.

Webreaper, 2025-01-16:
Yeah, I don't do any estimation at all. Only use the real rates.

Does look like it might be a rounding thing, but will investigate.

dqj999, 2025-01-16:
Not sure if this is the problem or not but when I went into Settings the "Always charge below" rate had been set to15 and even if I set it to 20 it would revert back to 15. 
Here's a link to a screenshot . [Screenshot](https://drive.google.com/file/d/1CR6Yz4NMkQ6z7GU99N4K4v-lLpvUOUl-/view?usp=drive_link)

Webreaper, 2025-01-16:
Nice spot. I suspect I've set the range of that value from 0-15p, thinking about agile fixed export. Didn't consider your use case. Should be an easy fix.

dqj999, 2025-01-16:
Thanks Mark.
BTW I'm still working on the version from 2 days ago. I'll update again
tomorrow.
Derek

Webreaper, 2025-01-16:
Should have a fix version out sometime tomorrow. 

In the meantime, if you manually edit the config JSON file, it should work so you can test it out - as long as you don't click 'Save Config' settings again.

closed 2025-01-16 by Webreaper

Webreaper, 2025-01-16:
Released the fix for this.

commit be0801a6 (2025-01-16): Fix #46