Webreaper/SolisAgileManager #56: Solcast API Calls

HarryFlatter, 2025-01-21:
I currently have 2 arrays configured in Solcast and use it primarily in a Home Assistant automation and dashboard display. 

I update the dashboard 5 times a day and so that is 10 calls to the API (5 for each array). Since we are limited to 10 API calls per 24 hrs , this means Solis Monitor pinches some of my Home Assistant calls.

I could probably reduce my Home Assistant updates to 3 times per day (6 API calls), so could Solis Monitor survive with 4 calls to the API? It would be up to 4 calls for a single array or 2 calls for twin arrays. 

Would it make sense to parameterise this in the config file?

Webreaper, 2025-01-22:
Okay, next version will have an additional config setting. By default the app will just do one call at 02:13am, but there'll be an option (defaulting to 'off') to have extra calls at 6am and midday, to improve forecasting accuracy, if you have the spare API rate-limit availability.

Webreaper, 2025-01-22:
(Still have to work on the ability to specify multiple solcast API site keys - #52)

HarryFlatter, 2025-01-22:
Thanks for the update :-).

Since many overnight 'cheap' rates commence at just gone midnight, would it make sense to make that first call at 00:13 so that when you are utilising the forecast data, the app can decide whether or not to action a battery charge based on forecasted solar yield for later that day?

closed 2025-01-28 by Webreaper