Webreaper/SolisAgileManager #66: Easy method of checking installed version

dqj999, 2025-01-25:
In creating a bash script to update the raspi version of SolisManager, I needed to check what version was currently installed. I couldn't find an easy way of doing that (there may be one - I don't know). In the end I created a workaround by writing a file whose name was the version number that had just been installed into a version subdirectory of the installation. If the version currently installed matches the planned installation version then the script exits.
Some  record of the current version that could be easily extracted with a bash script would be helpful.

Webreaper, 2025-01-25:
I can update the build to write the version number out to a file and
include it in the release.

Webreaper, 2025-01-28:
Okay, I've just updated the build - the packages that are built will now contain a file called version.txt with the build version in it. 

See this dev build for an example: https://github.com/Webreaper/SolisAgileManager/actions/runs/13005103228#artifacts

closed 2025-01-28 by Webreaper