Webreaper/SolisAgileManager #68: Price labelling in Charging Plan

dqj999, 2025-01-25:
I'm on Octopus Go which has fixed rates (27.10p/unit) from 5:30 to 00:30. I noticed in the Charging Plan that 15:30 to 19:00 was labelled as "Peak" and the rest of time as "Average". 

What causes this difference in the labelling, is it the result of some history of prices?  I know that on Agile these would indeed tend to be Peak charge times.

Webreaper, 2025-01-25:
I might make an adjustment to make this more intuitive: once the peak period has been found, look for adjoining slots at the same rate or higher than the peak average, and mark them as peak too.

This might also solve #40

Webreaper, 2025-02-27:
This is done in the latest release.

closed 2025-02-27 by Webreaper