Webreaper/SolisAgileManager #69: Solar PV Forecast Today is much less than the sum of half-hour forecasts

dqj999, 2025-01-27:
Today the Solar Forecast Today label was predicting 0.8Kw, but the half hour forecast PV figures were predicting about 0.25 kWh every half hour over about 8 slots.  Is the slot forecast meant to be the rate of power generation per hour or the amount of power estimated to be generated in the half hour? 

Perhaps one of these has the damping factor applied and not the other.

Incidentally it is not clear what the damping factor does to the estimated solar power generation. If 10Kw of solar power was predicted and the damping factor was 25%, would that mean the corrected estimate was 7.5Kw (percentage reduction) or 2.5Kw (multiplier). The 50% example unfortunately gives the same result for both methods.

Webreaper, 2025-01-27:
In the latest dev version this is all much-improved; the last release version doesn't have the damping factor applied and there's a couple of other bugs. 

The damping factor might be counter-intuitive - it basically means "use this percentage of the solcast figures". So if you set 80%, then it'll reduce the solcast estimate by 20%. Again, I can probably make this more intuitive, but I'm re-writing it so it's calculated automatically anyway (see #65) - once that's working, you won't need to manually set the damping factor at all (but I might leave it there as a manual override).

dqj999, 2025-01-27:
I get it - so it's like a de-rating factor. 

BTW Brilliant piece of software. Ticking over very nicely on my 64-bit Pi 3B.

Webreaper, 2025-01-27:
Glad you're liking it. Just pushed a new release with better config set up (uses the Octopus API key and account number) plus a bunch of other new figures. See the release for details

Webreaper, 2025-01-28:
I've found a fix for this issue - one of my calculations was off. Will push a release once some other stuff is ready.

Webreaper, 2025-01-31:
This should be fixed in the next release.

closed 2025-01-31 by Webreaper