Webreaper/SolisAgileManager #72: Comparison graphs have no vertical axis labelling

dqj999, 2025-01-28:
The Comparison graphs have no vertical axis labelling.

Webreaper, 2025-01-28:
They do for me? Can you provide a screenshot.

[image]

dqj999, 2025-01-28:
[image]

I wonder is this is a light/dark scheme problem?

dqj999, 2025-01-28:
Shift Refresh didn't work. 
I'm getting the same results with Chrome, Edge, Firefox, Safari

The error  seems to be that the number text has the property fill="white" on my Chrome browser.  Changing it to fill="blue" made the number visible. 

[image]

Webreaper, 2025-01-28:
Oh, wait, you're talking about the actual Axis, aren't you. Got it, yes, you're right, something's up.

Webreaper, 2025-01-28:
Okay, thanks. Found it, that'll be fixed in the next build.

closed 2025-01-31 by Webreaper