Webreaper/SolisAgileManager #74: Charging History Display - Show time as well as day when mousing over a slot

dqj999, 2025-01-29:
In the Charging History Display,  as you move the mouse over the display, there is a pop up that shows the Price and the SOC. 

The top line of the pop up shows the date. It would be useful if it showed the start time of the slot as well.

Webreaper, 2025-01-29:
Do you mean in the graph? If so, yes, I haven't worked out how to customise that floating popover in the Apex Charts library yet, and this is something that's required. :)

dqj999, 2025-01-29:
Yes on the graph. The Charts are pretty amazing anyway!

dqj999, 2025-01-29:
Have you tried using Google AI Studio for solving issues? I'm playing with the (currently) latest version which is Gemini 2.0 Flash Experimental that has the added advantage of being free! 
I stuck in a query about "Customise Floating PopUp in  Apex Charts library" and got some useful info and then I asked it about displaying time and date instead of just date. The response is attached.

[Google AI Studio Response.txt](https://github.com/user-attachments/files/18589691/Google.AI.Studio.Response.txt)

Apologies if this is in the egg-sucking info department!

Webreaper, 2025-01-29:
Yeah, no thanks. AI queries like that are for lazy developers. The problem with LLMs is that they frequently hallucinate, or return solutions which are stale or not idiomatic, and often sub-optimal. I prefer just finding a sample or SO post, which actually answers my query. 

The reason I haven't fixed the graph popovers isn't because I can't find the answer, it's because I've had way more important things to focus on. It'll take 10 minutes to fix when I actually have the time and bandwidth, but I'm working on the whole Solcast thing and so have bigger fish to try at the moment. Plus my day job. 😄

Webreaper, 2025-01-29:
Nice! 

And yes, worth having the issue raised, so when I'm bored (Ha!) I can fix it.

dqj999, 2025-02-17:
If the mouseover showed all the info there would be no need to add Vertical axis labels for the solar values, making it easier to see what was happening. Just a thought.

Webreaper, 2025-02-17:
Yep, that's on the plan.

Webreaper, 2025-02-27:
I marked it as closed when I committed the code. I've just done a new release which has this in it.