Webreaper/SolisAgileManager #79: Add label showing predicted Solar power for tomorrow

dqj999, 2025-01-30:
As well as today's solar power prediction it would be useful to have the estimate of tomorrow's solar power (when available). 
This would make it easier to manually adjust the amount of power to be added to the battery overnight.

Webreaper, 2025-01-30:
It's coming.... If you want to try a preview, download and update to the artifacts in this build when it's complete:

https://github.com/Webreaper/SolisAgileManager/actions/runs/13057697584#artifacts

There's still a small question about the forecast calculation, so I'd be interested to see what you get.

dqj999, 2025-01-31:
Having an issue downloading that to my Pi 3. For some reason the Pi can't find the zip file.

wget https://github.com/Webreaper/SolisAgileManager/actions/runs/13057697584/artifacts/2512151010
--2025-01-31 16:42:20--  https://github.com/Webreaper/SolisAgileManager/actions/runs/13057697584/artifacts/2512151010
Resolving github.com (github.com)... 20.26.156.215
Connecting to github.com (github.com)|20.26.156.215|:443... connected.
HTTP request sent, awaiting response... 307 Temporary Redirect
Location: https://github.com/Webreaper/SolisAgileManager/suites/33733717234/artifacts/2512151010 [following]
--2025-01-31 16:42:20--  https://github.com/Webreaper/SolisAgileManager/suites/33733717234/artifacts/2512151010
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 404 Not Found
2025-01-31 16:42:21 ERROR 404: Not Found.

Webreaper, 2025-01-31:
Yeah, you'd probably have to do it manually for a dev build. I'll probably push a release over the weekend.

Webreaper, 2025-01-31:
If you still want to manually install a dev build, this has quite a few fixes around the solcast stuff. https://github.com/Webreaper/SolisAgileManager/actions/runs/13076900581

closed 2025-01-31 by Webreaper