Webreaper/SolisAgileManager #81: Set battery percentage to force charge

itolson, 2025-01-31:
When discharging the batteries set the minimum state of charge percentage so that it dras from the grid at that battery percentage.

there could also be a force charge percentage level what ever the easiest way to implement it would be i guess.

Great front end, looking forward to some new features.

Webreaper, 2025-01-31:
Just so I understand, are you requesting a setting so if the battery drops below a certain percentage, it charges? So you can say something like "never let the battery drop below 30%"?

itolson, 2025-01-31:
Hi thanks for the great work. I'm trying to maximise profits on exporting
as importing is 7p and export 15p means a credit on the bill  so at the end
of the day the batteries are around 80 percent at 10 o'clock so need to
discharge until 23:30 when they can charge again.

Problem is to automate if the batteries are less then the percentage set on
the minimum SOC x percent it stops discharging and either uses the Grid.
The original solis setting was the force charge at x percent setting.

It's so close to being perfect.

itolson, 2025-01-31:
Ps will buy you a couple of coffee's

Webreaper, 2025-01-31:
So what you really want is #83 combined with this - i.e., "_between 22:00 and 23:30, discharge the battery until it's at 30%, and then stop discharging_"?

itolson, 2025-01-31:
Between x and y time to discharge  until it's at x percent battery percent.
x and y plus battery percentage being a setting to choose

closed 2025-02-04 by Webreaper

itolson, 2025-02-04:
Great!