Webreaper/SolisAgileManager #83: schedule an mode

itolson, 2025-01-31:
Schedule a reoccurring action ie discharge at 22:00 for 2 hours currently you need to do this manually every day.

Webreaper, 2025-01-31:
This might be getting a bit far outside the original design, but it's probably not too complex to do. Not sure when I'll be able to consider it, but I'll have a think.

Can you explain a bit more about the use case - ie how and why you want to discharge at that time every day?

itolson, 2025-01-31:
I've scheduled the discharge from 22:00 to 23:30 but need to stop if the
battery percentage goes below x percent so change the plan to no action.

itolson, 2025-02-01:
I think the force charge should be ok as going onto the control invertor option on the SOLIS cloud shows it as the percentage I set it at.

The schedule thing would be great if you can sort it.

Thanks

Webreaper, 2025-02-01:
I've added the "charge if SOC below certain level" setting. If you pull the
dev tag for the container you can try it.

I started building the schedule thing (it's really a "daily manual override
per slot" setting). Should be ready sometime next week.

Thanks for the coffees!

Webreaper, 2025-02-02:
Okay, you can test this out now. UI needs tidying up, but it's working.

Use the `webreaper/solisagilemanager:dev` image, and on the config page you can add a scheduled action for each slot. Note that currently there's no validation, so if you add two entries for the same slot with different actions, chaos will ensue. I'll add the validation tomorrow. 

The "always charge if SOC is below x" setting will take precedence over the scheduled actions, so you should be able to achieve what you need. Note that currently this is checked at the start of every 30 minute slot - so if you set it to charge below 30%, and at the start of the slot it's 31%, it won't charge until 30 mins later. I'll fix that later in the week.

itolson, 2025-02-02:
Hi no worries I'll try it and let you know thanks again

itolson, 2025-02-02:
Hi the minimum soc I can set but the schedule does not seem to do anything
I save and refresh then go back to schedule but it's empty. Let  e know if
you want me to test another version. No rush though. Cheers

Webreaper, 2025-02-02:
Did you save the config after adding the schedule? The 'save button' in the
popup is really 'close', you still need to click save config after it's
closed

itolson, 2025-02-02:
Hi tried again it's now saving but not updating the list was that to be
done later?

Webreaper, 2025-02-02:
Updating which list?

What scheduled action did you set? Currently it won't appear until the time reaches the actual slot where the scheduled action is set. Again, I'll have it update the future plan in the next iteration.

itolson, 2025-02-02:
Ok sorry I understand now it won't change until it is in that time slot. I
will give it a go thanks again

Webreaper, 2025-02-02:
To be fair, I should have the UI update immediately. 😁

itolson, 2025-02-02:
Haha, thanks!

Webreaper, 2025-02-02:
Okay, just pushed a bunch of code changes to the `webreaper/damselfly:dev` tag.
- Validation on the scheduled actions popup (in case you try and create two scheduled actions with the same time)
- Charging plan will now show future scheduled actions (with a little clock to indicate they're scheduled overrides)
- Config screen has a more meaningful description and will prompt if you make changes but haven't clicked 'Save Config'
- Some other stuff... :)

The only thing not working now is that if you have a scheduled action, and click the 'manual override' icons to override the scheduled action as a one-off, it doesn't update. I'll fix this tomorrow.

Webreaper, 2025-02-02:
Build is here: give it 10 mins: https://github.com/Webreaper/SolisAgileManager/actions/runs/13100729617

itolson, 2025-02-02:
Hi Great I'll give it a go cheers

Webreaper, 2025-02-02:
> Hi maybe doing something wrong but there an error when saving don't worry
> about. It's the weekend lol.

What's the error? Kinda hard to diagnose without the error message....

itolson, 2025-02-02:
Sorry I thought I'd attached it's in the log fire the container

Webreaper, 2025-02-02:
I don't see an attachment. Can you just past the error from the logs into a
comment here?

itolson, 2025-02-02:
Yes sorry it's solis cloud 403 forbidden I'll check my key and secret

itolson, 2025-02-02:
Says error posting request to solis cloud

Webreaper, 2025-02-02:
Weird. Wonder if there were too many requests or something?

itolson, 2025-02-02:
Have run the old version for now I'll test the later tomorrow. Have a good
evening and thanks again.

Webreaper, 2025-02-02:
I've changed nothing with the Solis API code so can't see how you'd get a
403 from the dev version...

itolson, 2025-02-04:
re-deployed the docker image and its working.

Webreaper, 2025-02-04:
Nice. Thanks.

closed 2025-02-04 by Webreaper