Webreaper/SolisAgileManager #86: Charging History Graph does not show price info on narrow screens

dqj999, 2025-02-01:
When using  a narrow screen, such as on a mobile phone, the Charging History Graph does not show the charging price, (probably because the line widths get below a small amount). You can see this effect on a PC screen if you artificially reduce the screen width available to the browser.

[image]

At slightly larger widths the charge type colour disappears.

[image]

Webreaper, 2025-02-01:
Yep, been looking at that. Might need to convert to an area graph. In the process of adding forecast and actual PV generation, and generally tidying up this graph.

dqj999, 2025-02-01:
I think  that this is made worse by the gaps and border between cells - presumably that doesn't occur with the other type of graph.