Webreaper/SolisAgileManager #87: Help information on Pricing & Charging Plan Screen cannot be displayed on touch screen device

dqj999, 2025-02-01:
If you hover the mouse cursor over any of the labels at the top of the screen, extra information is displayed e.g. "Current load being consumed by the house".

This doesn't happen on a touch screen device, and currently there is no way to see this info. On the Config Screen there are specific Help Icons that you can click to get more information.

Webreaper, 2025-02-01:
Tool tips don't generally work on touch screen device. Hopefully people will learn what the icons mean so won't need them on a phone.

One thought I had is to duplicate the colours and icons of the Soliscloud app so people will be familiar.

Webreaper, 2025-02-01:
One thing I could do is make them tappable or clickable, and display a popup with the description, bit like the `?` icons in the config screen.

dqj999, 2025-02-01:
Making them clickable would work :-)

The pair I find confusing are the "Total Solar PV Generation Today" and the "Current Solar PV Generation".  Maybe putting a Sigma in front of it would help the mathematically oriented amongst us!

Webreaper, 2025-02-01:
Doesn't the inclusion of the units make it absolutely clear?

dqj999, 2025-02-01:
Yes it does if I think about it! 

Doh!

Webreaper, 2025-02-04:
Fixed in most recent release.

closed 2025-02-04 by Webreaper