Webreaper/SolisAgileManager #88: Solar Power Estimates for each slot stop being displayed after active usage

dqj999, 2025-02-01:
After using the App for some time I noticed that the column for Solar Power Estimates has vanished from the Prices and Charging Plan Display. The forecast PV was also showing as Zero.

[image]

I think that this probably a consequence of inadvertently over-using the SolCast data forecast with a number of restarts and refreshes; so probably not an issue that most people would experience.

(Thought that I would mention it just in case the cause is something else)

dqj999, 2025-02-01:
This was caused by restarting the App. 

Last SolCast update recorded as "Never" 

No issue here!

Webreaper, 2025-02-01:
Have completely rewritten the Solcast caching so all these issues should be solved in the next release, and you can restart as often as you like.

closed 2025-02-03 by Webreaper