Webreaper/SolisAgileManager #89: Charging strategy - Saving battery to avoid peak usage

dqj999, 2025-02-02:
When using a tariff with high peaks in the 4pm to 7pm period and extreme lows overnight, sometimes the best option for me is to put the inverter into Hold mode during the pre-peak period where it will not discharge or charge. (This happens when you set the inverter to charge at 0 Amps.)
On low sun days my battery would not last till 4pm even if fully charged by 6am, so doing this is reduces the overall cost.

To implement this it would need another state, “Hold”, which could just be manually selected for particular slots if there was no automatic provision of this state.

dqj999, 2025-02-03:
If you could just split the charge rate and discharge rate in the config and allow it to go down to zero then I can manually adjust things on days which might benefit from holding the charge in the battery.

Webreaper, 2025-02-03:
I don't want to build the same UI as the (terrible) Solis Inverter control API. 

I think what you want is already there now, suggested by somebody else:

- I need to add a new 'Hold' action, where it sets a discharge slot with zero amps, to prevent the battery being used
- There's now the concept of 'Scheduled Actions', where you can schedule an action to happen every day (e.g., always discharge the battery for 90 minutes at 8pm)
- There's a new setting in the latest version 'Charge below SOC %` which will charge the battery regardless of price to ensure the battery remains above that percentage.

I can add the 'hold' Action quite easily.

Webreaper, 2025-02-03:
Try this version out - it has the hold action implemented. I haven't tested this at all, but it should work. :)

https://github.com/Webreaper/SolisAgileManager/actions/runs/13119467396

Let me know how it goes - planning to do a release in the next couple of days.

dqj999, 2025-02-03:
I'll try that out asap.

Could you tell me where I find the raspi.zip file from that build link please?

Webreaper, 2025-02-03:
Click the "artifacts" link towards the top right

Webreaper, 2025-02-04:
Released

closed 2025-02-04 by Webreaper