Webreaper/SolisAgileManager #94: Charging Plan - Left align Forecast column data

dqj999, 2025-02-04:
In v1.0.340.0

Webreaper, 2025-02-04:
I just changed it to right align. Why would we want it left aligned?

dqj999, 2025-02-04:
It looks a bit odd with a large blank area to the left, and it is not under the Heading. (Centered, Narrower column?)

[image]

Webreaper, 2025-02-04:
Make your browser less wide. 🤣

Webreaper, 2025-02-04:
I think the actual solution to this is simpler: 

1. Swap the forecast and 'Manual Actions' columns
2. Limit the max-width of the forecast column

The other advantage that brings is that the forecast data etc would all be together. Something like this:

[image]

dqj999, 2025-02-04:
I like that format.👍

Webreaper, 2025-02-05:
Released!

closed 2025-02-05 by Webreaper