Webreaper/SolisAgileManager #96: Slight tweak to the Charge if SOC (%) Below.

itolson, 2025-02-04:
There is a Always charge below percentage option so setting this will stop the discharging and force charge.

Can the Charge if SOC (%) Below be changed to change that timing period status to do nothing or the next period as long as its not within the cheap charge period ?

just after a set and forget automation if away etc to discharge the battery at x oclock which is in the scheduled and then stop charging if the battery SOC  goes below the parameter value.

dqj999, 2025-02-04:
I have a use case where the "Always Charge Below %" needs to be unconditional so that I can control the situation where the Solar Battery may inadvertently be used to charge my EV battery.

itolson, 2025-02-04:
I use HA to control the Zappi to only charge off peak 23:30 and then set to stopped at 05:30, I discharge the solar batteries 25kwh after 21:00 until the charge at 23:30 the scheduled cannot be overridden manually so i had to change the schedule to stop the discharge.

Webreaper, 2025-02-04:
I don't really understand what is being asked for here....

> Can the Charge if SOC (%) Below be changed to change that timing period status to do nothing or the next period as long as its not within the cheap charge period ?

Isn't this just a scheduled action of 'Charge if low battery' and then set the low battery threshold to the desired %age?

Webreaper, 2025-02-04:
Thinking about it, would it solve the problem if you could do this:

[image]

...but with an explicit percentage specified for that action? And add the limit for other scheduled actions? 

So the options for a scheduled action would be: 
* **Charge** - but optionally only if price is below `x`
* **Discharge** - but only until the battery is at %age `x`
* **Charge if Low Battery** - where 'low battery' is defined as part of the scheduled action, not the global %age

Would that work?

itolson, 2025-02-04:
Hi, yes that would work, as you say keep things simple, it starts to get more complex when your scheduling discharges and maximising export profits.

martinpike69, 2025-03-05:
Re the variable charging rate, another option would be to add a 'duration' box, say X 'hours' or 'slots' (of 30 mins), below which you wish to charge at Y rate (say 100 amps, splash and dash), and above which you are happy to charge at the standard rate, say 75 amps?  Just a thought.  At the end of the day, whatever you around this, it's adding complexity.

Webreaper, 2025-03-05:
Okay, latest dev build has the ability to override the global amps setting for scheduled actions. I need to tidy up the UI but for charge / discharge actions you'll get a new box for "amps". If you leave it blank it'll use the global setting, otherwise it'll use the value specified.

Give it a try and let me know if it works. I'll tidy the UI tomorrow.

martinpike69, 2025-03-05:
Sorry, where do I download the latest dev build from please?

Webreaper, 2025-03-06:
@martinpike69 - you can always get the latest buidl from https://github.com/Webreaper/SolisAgileManager/actions - click on the top entry in the grid, and then click on the 'artifacts' link in the top right.

Be aware that sometimes the dev build may have in-progress test code which isn't 100% functional though, so use at your own risk.

Webreaper, 2025-03-06:
Tested this today - currently doing a release.

closed 2025-03-06 by Webreaper