Webreaper/SolisAgileManager #97: History Grid View on narrow screens

dqj999, 2025-02-04:
On narrow screens, such as a mobile phone in Portrait view the Grid View in History is not useful.
Perhaps this option should be supressed when the available width is below some threshold. 
An option to list the data in CSV form might be a useful solution here.

[image]

Webreaper, 2025-02-04:
Yes, this is the default behaviour of the grid I use when the screen is narrow. I need to work on it. 

The data is already in the ExecutionHistory.csv file in the config folder (that table is basically a grid view of what's in the csv file). I suppose I could add a download link

Webreaper, 2025-02-04:
Improvements that I think this needs: 

1. Virtualisation to make it faster
2. Summary rows for each day
3. Better formatting
4. Column Header sorting

Plus other tidy-ups

Webreaper, 2025-02-05:
Some of these are in the latest release.