Webreaper/SolisAgileManager #98: Rename "No Action" to "Auto"

dqj999, 2025-02-04:
If you had limited familiarity with the Solis Inverter, the concept of "No action" would seem a bit strange. We know that it means "Self Use" in Solis terms, so either calling it that or "Auto" would seem to be an improvement.

Webreaper, 2025-02-04:
I like Auto. Maybe 'self use' is better. That said, it's describing the plan for the app, so I'm torn. One to think about.

If you have any better suggestions for the icons please send them my way too. I'm still unconvinced about the 'No Action' and 'Hold' icons.....

dqj999, 2025-02-04:
How about rotating the battery discharging  icon to horizontal . Possibly with some red liquid in the battery pouring out

Webreaper, 2025-02-04:
I'm not bloody Pixar!!! 🤣

dqj999, 2025-02-05:
Some possible icons for Hold, Charge and Discharge

[image]
[image]
[image]

<a target="_blank" href="https://icons8.com/icon/96229/empty-battery">Battery</a> icon by <a target="_blank" href="https://icons8.com">Icons8</a>