Webreaper/SolisAgileManager #103: issue when opening the web page after a few hours.

itolson, 2025-02-07:
the charge plan does not refresh the previous time I closed the web page it seems like an overnight thing.

The charging plan overview was left at the state that I closed the web page down the previous day.

I then stopped and started the container and it looked like the attached.

[image]

The only way I can fix it is to redeploy a new docker image and it works again for x amount of time.

any idea's ?

Webreaper, 2025-02-07:
No idea. I've never seen this. 

Have you mapped an external volume so the config and other cache files for the history etc are written outside the container?

itolson, 2025-02-07:
found this in the container log 
[08:51:48.001--ERR] Unexexpected error recalculating slot plan
System.InvalidOperationException: Sequence contains no elements
   at System.Linq.ThrowHelper.ThrowNoElementsException()
   at System.Linq.Enumerable.Max[TSource,TResult](IEnumerable`1 source, Func`2 selector)
   at SolisManager.APIWrappers.SolcastAPI.get_lastAPIUpdate() in /src/SolisManager/APIWrappers/SolcastAPI.cs:line 17
   at SolisManager.Services.InverterManager.EnrichWithSolcastData(IEnumerable`1 slots) in /src/SolisManager/Services/InverterManager.cs:line 31
   at SolisManager.Services.InverterManager.RecalculateSlotPlan() in /src/SolisManager/Services/InverterManager.cs:line 249
[08:52:06.708--INF] Application is shutting down...
[08:52:06.725--INF] Coravel's Scheduling service is now stopped.
Config folder set to /appdata.
[09:32:17.746--INF] ===========================================================
[09:32:17.797--INF] Application started. Build version v1.0.349.0 Logs being written to /appdata

itolson, 2025-02-07:
may have been the solcast api key expiration.

Webreaper, 2025-02-07:
Thanks. Will fix.

Webreaper, 2025-02-07:
Pick up the latest dev tag in about 10 minutes and let me know if that's better.

Webreaper, 2025-02-07:
BTW, you mention solcast key expiration - I'm not aware that it has an expiration date?

itolson, 2025-02-07:
not sure but it started to work when i regenerated the api key and updated the config and it started to work.

closed 2025-02-11 by Webreaper