Webreaper/SolisAgileManager #104: [14:00:13.663--ERR] ERROR returned when reading inverter charging state

itolson, 2025-02-10:
When this happens the software thinks the battery is Zero and changes the action to charge, if you go into config and then save it then retrieves the correct battery percentage.

How often does it get the information from the inverter ?

Webreaper, 2025-02-10:
It gets the info from the inverter every 5 minutes. 

Why is it failing - are there errors in the log? If the inverter call fails, it should just skip that one - I'll check the logic. Mine has never failed on this call though!

Webreaper, 2025-02-10:
Actually, ignore the above answer. The error you're seeing is when it checks to see what is the state of the 'charging slots' in the inverter, so it knows whether it needs to do something or not. So for example, if we're about to apply a charge for the next 90 minutes, and the inverter charging slots are _already_ set to charge for the next 90 minutes, we do nothing, as no change is required. 

I'm not sure why this would affect anything. If the inverter doesn't return the charge state, it'll definitely apply the new charging slot, to ensure the right action is executed. But this won't change what the inverter is doing. 

If the calls to the inverter to get the battery SOC are failing, that's something else entirely - but that's not what the error message at the top of this issue states.

Webreaper, 2025-02-10:
When I check my logs I have one error calling the Solis API, but you can see the SOC doesn't drop to zero after the failed call.

```
[07:40:00.650--INF] Refreshed state: SOC = 89%, Current PV = 0.066kW, House Load = 2.228kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[07:45:00.763--INF] Refreshed state: SOC = 88%, Current PV = 0.064kW, House Load = 2.603kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[07:50:00.642--INF] Refreshed state: SOC = 87%, Current PV = 0.075kW, House Load = 2.164kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[07:55:00.793--INF] Refreshed state: SOC = 86%, Current PV = 0.077kW, House Load = 2.284kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[08:00:01.540--INF] Found Octopus Product/Contract: E-1R-COSY-22-12-08-J, Starts "2025-01-21T00:00:00.0000000Z"
[08:00:01.953--INF] Refreshed state: SOC = 83%, Current PV = 0.077kW, House Load = 6.296kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[08:00:02.747--INF] Retrieved 5 rates from Octopus (10-Feb-2025 22:00 - 10-Feb-2025 13:00) for product E-1R-COSY-22-12-08-J
[08:00:07.120--INF] Skipping charge request (Inverter state matches: 0, 0, 00:00-00:00, 00:00-00:00)
[08:00:07.121--INF] Enriching history with PV yield for 1 days
[08:00:07.384--INF] Retrieved 96 inverter stats for 10-Feb-2025
[08:06:40.506--ERR] Error posting request to SolisCloud: /v1/api/inverterDetail {"sn":"6031050232030275"}
System.Threading.Tasks.TaskCanceledException: The request was canceled due to the configured HttpClient.Timeout of 100 seconds elapsing.
 ---> System.TimeoutException: A task was canceled.
 ---> System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)
   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   --- End of inner exception stack trace ---
   --- End of inner exception stack trace ---
   at System.Net.Http.HttpClient.HandleFailure(Exception e, Boolean telemetryStarted, HttpResponseMessage response, CancellationTokenSource cts, CancellationToken cancellationToken, CancellationTokenSource pendingRequestsCts)
   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)
   at SolisManager.APIWrappers.SolisAPI.Post(String url, String content) in /src/SolisManager/APIWrappers/SolisAPI.cs:line 409
[08:06:40.509--ERR] Unexpected exception during inverter state refresh
System.Text.Json.JsonException: The input does not contain any JSON tokens. Expected the input to start with a valid JSON token, when isFinalBlock is true. Path: $ | LineNumber: 0 | BytePositionInLine: 0.
 ---> System.Text.Json.JsonReaderException: The input does not contain any JSON tokens. Expected the input to start with a valid JSON token, when isFinalBlock is true. LineNumber: 0 | BytePositionInLine: 0.
   at System.Text.Json.ThrowHelper.ThrowJsonReaderException(Utf8JsonReader& json, ExceptionResource resource, Byte nextByte, ReadOnlySpan`1 bytes)
   at System.Text.Json.Serialization.JsonConverter`1.ReadCore(Utf8JsonReader& reader, T& value, JsonSerializerOptions options, ReadStack& state)
   --- End of inner exception stack trace ---
   at System.Text.Json.ThrowHelper.ReThrowWithPath(ReadStack& state, JsonReaderException ex)
   at System.Text.Json.Serialization.JsonConverter`1.ReadCore(Utf8JsonReader& reader, T& value, JsonSerializerOptions options, ReadStack& state)
   at System.Text.Json.JsonSerializer.ReadFromSpan[TValue](ReadOnlySpan`1 json, JsonTypeInfo`1 jsonTypeInfo)
   at SolisManager.APIWrappers.SolisAPI.Post[T](Int32 apiVersion, String resource, Object body)
   at SolisManager.APIWrappers.SolisAPI.InverterState() in /src/SolisManager/APIWrappers/SolisAPI.cs:line 46
   at SolisManager.Services.InverterManager.UpdateInverterState() in /src/SolisManager/Services/InverterManager.cs:line 712
[08:10:00.775--INF] Refreshed state: SOC = 77%, Current PV = 0.079kW, House Load = 7.464kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[08:15:00.652--INF] Refreshed state: SOC = 74%, Current PV = 0.076kW, House Load = 7.805kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[08:20:00.812--INF] Refreshed state: SOC = 71%, Current PV = 0.079kW, House Load = 8.076kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[08:25:00.631--INF] Refreshed state: SOC = 69%, Current PV = 0.081kW, House Load = 1.646kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
[08:30:00.751--INF] Refreshed state: SOC = 69%, Current PV = 0.082kW, House Load = 2.173kW, Forecast today: 4.11570kWh, tomorrow: 5.90685kWh
```

itolson, 2025-02-10:
It has only dropped to zero twice this week. restarting the container or going into config and save seems to sort it.

I get this in the log before the error 

16:39:02.340--INF] Now listening on: http://[::]:5169
[16:39:02.341--INF] Application started. Press Ctrl+C to shut down.
[16:39:02.341--INF] Hosting environment: Production
[16:39:02.341--INF] Content root path: /app
[16:39:02.447--INF] Loaded 189 entries from execution history file SolisManagerExecutionHistory.csv
[16:39:06.759--INF] Retrieved 4 rates from Octopus (11-Feb-2025 23:30 - 10-Feb-2025 23:30) for product E-1R-INTELLI-VAR-22-10-14-L
[16:39:06.980--INF] Refreshed state: SOC = 85%, Current PV = 0.143kW, House Load = 0.779kW, Forecast today: 0kWh, tomorrow: 0kWh
[16:39:06.985--INF] 74 new Octopus rates available to 12-Feb-2025 05:30 (cheapest: 7.00035p/kWh, peak: 25.945605p/kWh)
[16:39:09.105--WRN] Failed to determine the https port for redirect.
[16:39:15.123--ERR] ERROR returned when reading inverter charging state
[16:39:15.125--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
[16:39:23.218--INF] Enriching history with PV yield for 1 days
[16:39:23.532--INF] Retrieved 195 inverter stats for 10-Feb-2025
[16:39:23.978--INF] Found Octopus Product/Contract: E-1R-INTELLI-VAR-22-10-14-L, Starts 10/03/2023 23:00:00
[16:39:23.982--INF] Updating inverter time to avoid drift...
[16:40:04.849--INF] Refreshed state: SOC = 85%, Current PV = 0.097kW, House Load = 0.951kW, Forecast today: 4.59230kWh, tomorrow: 6.24225kWh
[16:45:04.963--INF] Refreshed state: SOC = 84%, Current PV = 0.091kW, House Load = 0.962kW, Forecast today: 4.59230kWh, tomorrow: 6.24225kWh

It does not always goto zero though thats the weird thing.

Webreaper, 2025-02-10:
Okay, you can ignore this: 
```
[16:39:15.123--ERR] ERROR returned when reading inverter charging state
[16:39:15.125--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
```
It's benign. That's just saying:

1. I need to set the inverter to 'Do nothing' - i.e., all zeros in the charging slots
2. I'm going to check what the inverter charge slots are currently set to to see if I actually need to set that
3. Oh, I didn't get a response back. Right, I'll set it explicitly then
4. Setting it to do nothing, which is what it was doing before

I'm actually going to change the ERR to a Warning, because the only downside with this scenario is that there might be an unnecessary write to the inverter storage.

Webreaper, 2025-02-11:
I'm going to close this, as I don't think it's an actual problem. 

The app logs out the battery SOC every 5 minutes; if you see it drop to zero unexpectedly please send me the logs and I'll investigate.

closed 2025-02-11 by Webreaper