Webreaper/SolisAgileManager #105: random charging issue

itolson, 2025-02-12:
[image]

Not sure whats happing here but for some reason the manager decided not to stop charging in the peak times after 5:30 I noticed at 7:30 and manually set to do nothing.

then again at 9:00 ish it change to charge again. I went into the schedule and then save and it seems to have gone back to normal.

This is the first time this has happened to be honest.

logs show 

[08:30:04.894--INF] Sending new charge instruction to Solis Inverter: 95, 0, 08:30-09:00, 00:00-00:00
[08:30:05.225--INF] Enriching history with PV yield for 1 days
[08:30:05.317--INF] Retrieved 10 inverter stats for 12-Feb-2025
[08:35:04.728--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:40:04.510--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:45:04.636--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:50:04.553--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:55:04.733--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[09:00:04.321--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[09:00:05.031--INF] Retrieved 2 rates from Octopus (12-Feb-2025 23:30 - 12-Feb-2025 23:30) for product E-1R-INTELLI-VAR-22-10-14-L
[09:00:05.266--INF] Sending new charge instruction to Solis Inverter: 95, 0, 09:00-09:30, 00:00-00:00
[09:00:05.526--INF] Enriching history with PV yield for 1 days
[09:00:05.616--INF] Retrieved 10 inverter stats for 12-Feb-2025
[09:01:20.229--INF] Updating slot action for 02/12/2025 09:00:00 to DoNothing...
[09:01:20.229--INF] Set override: 09:00 - DoNothing
[09:01:24.462--INF] Retrieved 2 rates from Octopus (12-Feb-2025 23:30 - 12-Feb-2025 23:30) for product E-1R-INTELLI-VAR-22-10-14-L
[09:01:24.725--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[09:01:24.978--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
[09:01:25.271--INF] Enriching history with PV yield for 1 days
[09:01:25.346--INF] Retrieved 10 inverter stats for 12-Feb-2025
[09:01:30.841--INF] Saving config to server...
[09:01:31.229--INF] Found Octopus Product/Contract: E-1R-INTELLI-VAR-22-10-14-L, Starts 10/03/2023 23:00:00
[09:01:31.306--INF] Retrieved 2 rates from Octopus (12-Feb-2025 23:30 - 12-Feb-2025 23:30) for product E-1R-INTELLI-VAR-22-10-14-L
[09:01:31.321--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[09:01:31.630--INF] Skipping charge request (Inverter state matches: 0, 0, 00:00-00:00, 00:00-00:00)
[09:01:31.630--INF] Enriching history with PV yield for 1 days
[09:01:31.743--INF] Retrieved 10 inverter stats for 12-Feb-2025

Webreaper, 2025-02-12:
Seems odd. I'll take a look.

itolson, 2025-02-12:
I clicked the reset button and it read the values, maybe solis api / cloud was the issue not sure

itolson, 2025-02-12:
this is in the history

[image]

itolson, 2025-02-12:
is it possible to put a little logic in around the previous reported battery percentage and the current one so it does not charge if its say dropped by x percent in the last x minutes or something like that as it went from 100% at 5:30 to 0 and then change to charge. could have been a one off I guess.

Webreaper, 2025-02-12:
Yeah, can have a look at that. Probably should ignore SOC of 0%.

Webreaper, 2025-02-12:
I've pushed a change which will ignore the "charge if below X%" and "always keep battery at X%" rules if the SOC is zero. Also added some logging.

Give it about 10 minutes and then pull the dev tag and see if that solves it.

Webreaper, 2025-02-12:
I already log that for all solis calls, so something else is going on

itolson, 2025-02-12:
ok i will check the logs is this the dev release or latest ?

Webreaper, 2025-02-12:
Dev - I haven't done a release for a few days.

Webreaper, 2025-02-12:
So the weird thing I don't understand is that if the Solis calls were failing, I'd expect to see detailed logs about the errors. So I can't understand how it can fail for 6 calls in a row here, but without and errors being logged. It's most odd.

```
[08:35:04.728--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:40:04.510--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:45:04.636--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:50:04.553--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[08:55:04.733--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
[09:00:04.321--INF] Refreshed state: SOC = 0%, Current PV = 0.000kW, House Load = 0.000kW, Forecast today: 10.42965kWh, tomorrow: 9.72820kWh
```

Do you have any ideas why your inverter might not be responding for those 6 minutes? Is it possible wifi issues or somthing?

I've added (yet) more logging to the latest build to see if we get a clue.

itolson, 2025-02-12:
Maybe ignore the 0 percent as the fall below x percent should override so
if soc =0 which do nothing

Webreaper, 2025-02-13:
That's what the latest code does. It basically says: "If the latest SOC is zero, it _must_ be wrong, so ignore it and leave the last SOC as-is until the next reading". You can see this - when the inverter returns zero (or doesn't return anything) the 'last battery update' time won't change. So this should solve the problem.

itolson, 2025-02-13:
perfect thanks

closed 2025-02-19 by Webreaper