Webreaper/SolisAgileManager #106: Charging when not needed

By-tor, 2025-02-12:
I noticed an of issue today (although it could be incorrect settings on my part as I'm only 5 days in)

The algorithm is setting my batteries to charge when they are already at 100%
This causes the load to be supported by the grid when it could be satisfied by the battery

Have I setup something incorrectly or is this a genuine issue?

[image]

Webreaper, 2025-02-12:
Which Tariff are you on, and what are your settings?

Webreaper, 2025-02-12:
If you can provide a screenshot of the strategy page so I can see what the charging plan is for the next 12 hours, that would be usesful.

Webreaper, 2025-02-12:
Oh, one other thing it could be: if you're on a tariff like Cosy, Go etc., where the 'troughs' are always low and the same, then the 'cheapest' slots will all be the same and will all be set to charge. This is by design - the idea is that when the price is at its lowest, the grid is as cheap as using the battery, so keep charging and supply the load from the grid. Otherwise by the time you get to the place where the price rises, the battery might only be half-full, and you'd end up needing power from the grid when it's not cheapest.

By-tor, 2025-02-12:
Current settings:

[image]

Is this what you mean by strategy page?

[image]

[image]

I'm currently on Octopus Agile tariff

Webreaper, 2025-02-12:
Thanks. One last thing - could you attach the ExecutionHistory.csv file from the logs folder, so I can see why it did what it did?

By-tor, 2025-02-12:
[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/18772373/SolisManagerExecutionHistory.csv)

By-tor, 2025-02-12:
[SolisManager-20250212.log](https://github.com/user-attachments/files/18772621/SolisManager-20250212.log)

Webreaper, 2025-02-13:
Okay, I've run through the data in simulation mode, and the problem is that your "slots needed to charge full battery" is probably set too high. Currently, you have it set to 3 hours, which means that SolisManager is looking for the cheapest 3-hour period and setting them all to charge. If the battery is full, it will reduce that, but it will always apply some charging, to ensure the battery is topped up in the cheapest time. So when I ran it, if I decreased that setting down to '1 hour' I ended up with 2-3 charge slots instead of 6. The other cheapest slots were set to 'Charge if low battery' which will boost the battery if it's low when prices are cheapest. 

So the solution is to bump that setting down to, say, 1.5 hours, and it shouldn't charge for so long. You can see the differences here: 

This image shows the charging plan graph with that setting set to 3 hours - you can see there's a solid block of blue and it's going to charge for the whole time.
[image]

This shows it set to 1.5 hours - you can see that some of the cheapest slots have been converted to 'Charge if low bettery', which won't charge at all if the battery is above 30% (per your 'Boost battery' setting).
[image]

In either case, if the battery is full, it should reduce charging or not charge at all - but it will favour charging if it's the cheapest slot, because it's better (financially) to take cheapest power from the grid and hold the battery's charge than to use the cheap power in the battery, but end up having the battery run out earlier later in the day (meaning you have to use more expensive grid power). 

As a general observation, from the graph you posted it does look like your battery will last all day and so your charging strategy really just needs to charge during the overnight period and.... that's it. In that case, if your battery will always get you through the peak price period without a mid-day charge, you could drop the `Battery % for peak period` right down, which means that SolisManager will basically do nothing other than charge overnight during the cheapest period - it won't hunt for any pre-peak cheap slots unless your battery is right down.

Hope this makes sense.

By-tor, 2025-02-13:
I had set it to 3 hours as it takes that to charge from 15% to 99% SOC at 60A (14% per half hour x 6 = 84% plus 15% minimum SOC = 99%)

Thanks for the guidance, I will make the changes and see how it behaves.

Webreaper, 2025-02-13:
Yep, let me know if it works. If not, we can consider other adjustments. :)

By-tor, 2025-02-18:
Now running Solis Agile Manager v1.0.393.0 with the config settings revised as above, but it's still initiating charge slots even when the batteries are at 100% SOC. 
As can be seen below 100% reached at 13:20 and still set to charge at 14:25. 
Thankfully, the sun was shining enough to fulfil the load requirements or it would have been taken from the grid.

[image]

Webreaper, 2025-02-18:
I can't debug this without more information. The soliscloud graph isn't that helpful. :)

Can you please send me:
1. The logs for today
2. The details of your settings (specifically the revised "slots to charge battery" setting, but all of the general charge settings would be useful. 
3. Ideally, if you can get me a screenshot of the charging plan and plan graph in the run-up to 14:30 tomorrow, that would also be useful. 

Again, remember that the goal of the app is to ensure you've got enough charge to get through the peak period. When I look at today's Agile prices, it jumps from 22p to 40p within 2 slots:

[image]

So the app is more focused on ensuring you have a full battery for the peak, not that you're using that battery in the run up to the grid. What is your "Battery % for peak period" set to? Perhaps you have it too high?

By-tor, 2025-02-18:
Log:

[SolisManager-20250218.log](https://github.com/user-attachments/files/18847936/SolisManager-20250218.log)

Execution history: (just in case)

[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/18847939/SolisManagerExecutionHistory.csv)

Options screenshot:

[image]

Don't have the charging plan for tomorrow yet but here is todays:

[image]

History for today:

[image]

Webreaper, 2025-02-18:
Thanks, that's useful. If you could send me a screenshot of tomorrow's charging plan sometime in the morning (about 9am ish, just so that it's got the plan for 2pm) that would be useful. 

I'll take a look at everything here and see if/what needs to be tweaked. Probably won't get a chance to look until tomorrow evening though.

Webreaper, 2025-02-19:
Okay, I've made some adjustments - the SOC wasn't being properly taken into account when calculating whether charging is necessary. This should fix it. 

Could you try running this build and tell me if it works better for you? https://github.com/Webreaper/SolisAgileManager/actions/runs/13408575012

I need to do some additional testing to make sure the overnight behaviour doesn't do anything weird, but it should work much better and the "% required for peak" setting will have a more significant effect.

By-tor, 2025-02-19:
Pre- change charging plan:

[image]

Webreaper, 2025-02-19:
Looks good - we'll see how it changes as you approach the cheap time (based on my simulation runs it'll change around based on SOC).

By-tor, 2025-02-19:
Post-changes to .399:

[image]

Webreaper, 2025-02-19:
Ah, that's the new version? Got it. Thanks.

By-tor, 2025-02-19:
I am going on holiday for a week tomorrow so I will leave this version running and let you know how it behaves when I get back.

tabannis, 2025-02-20:
I suspect i'm having this issue, but with the Flux tarrif. CSV and logs attached for info.

[SolisManager-20250216.log](https://github.com/user-attachments/files/18893679/SolisManager-20250216.log)
[SolisManager-20250219.log](https://github.com/user-attachments/files/18893678/SolisManager-20250219.log)
[SolisManager-20250220.log](https://github.com/user-attachments/files/18893677/SolisManager-20250220.log)
[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/18893676/SolisManagerExecutionHistory.csv)

I'll try the dev build too :)

closed 2025-02-27 by Webreaper