Webreaper/SolisAgileManager #107: Intelligent Go Tariff - Knowing when charging your EV in the day time

dqj999, 2025-02-13:
If you are on the Intelligent Go tariff there does not seem to be an API that will tell you when Octopus is giving you cheap rate to charge your vehicle. Consequently if you connect your vehicle in the day the system may decide to charge your EV battery at any time and unless you manually set your inverter to Hold or Charge you could end up charging your EV from your solar battery.

I would like to solve this problem using a CT to detect current going to the EV charger. This can be a relatively simple piece of kit and I’ve been testing ways of doing this. A simple robust way of interfacing this to Solis Manager would be for the device to touch a file in the config directory whenever charging was detected. If SolisManager could check the timestamp of this file say every 5 minutes it would know when charging was taking place and could arrange to put the inverter into charge mode for the current slot. 

If the CT hardware were to become disconnected or fail there would be no consequences because the timestamp of the file would not be updated and so the system would not continue to think that EV charging was taking place.

Is there a possibility of implementing this?

Webreaper, 2025-02-13:
The implementation you describe is not something I'd do - mainly because the number of people who would actually set up a CT clamp as you describe is almost nil. 

This is a question I've been wondering about though. For IOG, I presume that Octopus must manage these ad-hoc charging sessions in 30-minute slots (since that's the meter granularity). The main question is whether there's an API to query to find out if/when a cheap slot is being provided, and have that feed into the app. 

Let me have a dig around and find out what's available/possible.

Webreaper, 2025-02-13:
So after a quick look, it appears there is an octopus API to get the dispatches for IOG. I won't be able to test, obviously, so I might need to do some test iteractions/cycles with you if that's okay, once I've built something? The alternative is that you could give me your Account number and API key so I could test - but you might not trust me that much. :)

dqj999, 2025-02-13:
I’m not currently on the Intelligent Go Tariff but I’ll change back to it
now and let you know.

Derek

Webreaper, 2025-02-13:
Okay, so I've added some code that will attempt to query the dispatches and will log them out (it won't do anything with them yet). It's in this build: https://github.com/Webreaper/SolisAgileManager/actions/runs/13307063164

It's entirely speculative, so I've no idea if it'll work at all, but it could be a good start - particularly if it returns data. Once I have the response data from the logs, I can then set about doing something useful with it. 

So if you switch to IOG, and give it a try, let me know if you see anything in the logs (even if it's just errors...).

Webreaper, 2025-02-13:
Just making notes for my own memory - it looks like the `PlannedDispatch` object has a `meta` property which has two fields. If the `source` is `SMART_CHARGE` then it means a smart cheap-rate charge is being done - at which point the inverter should be set to immediately charge for that period. :)

Webreaper, 2025-02-13:
Okay, thanks for the key/account, that was helpful. I think I've got the functionality in place, but difficult to know as there's no dispatches for charging slots yet. 

But if you want to give this version a go:

https://github.com/Webreaper/SolisAgileManager/actions/runs/13308158872

It should query the IOG dispatches every 30 mins when the plan is calculated, and if there's any cheap IOG charging slots, it'll apply them to the inverter too. I've added a fair bit of logging so we can see if it does the right thing. 

Give it a go and we can see if it works - and iterate if not!

Webreaper, 2025-02-13:
Another note-to-self: polling every 30 minutes is probably not frequent enough, because slots can stop/start at any time. So probably need to check the IOG plan every 1 or 5 minutes.

dqj999, 2025-02-13:
I now have a charging session scheduled by Octopus for 03:30 to 04:47 AM on 14th Feb.

Webreaper, 2025-02-13:
Just tested, and I'm not seeing anything come through on the planned dispatches API. 🤔

dqj999, 2025-02-13:
These are the results that I emailed to you earlier:

[13:00:00.904--INF] IOG Tariff - attempting to retrieve dispatches...
[13:00:01.243--INF] Refreshed state: SOC = 19%, Current PV = 0.352kW, House Load = 0.352kW, Forecast today: 0.28195kWh, tomorrow: 3.70715kWh
[13:00:01.587--INF] Retrieved 4 rates from Octopus (14-Feb-2025 23:30 - 13-Feb-2025 23:30) for product E-1R-INTELLI-VAR-24-10-29-A
[13:00:02.206--WRN] No dispatches returned from the octopus API

Webreaper, 2025-02-13:
Yeah, it's weird, I'm seeing none come through too. I'm probably missing something - I'll have a rummage later. 

But it looks like it should be able to work once I figure this out, and without a CT clamp!

Webreaper, 2025-02-14:
Okay, I've tested this morning, and the response is coming back with some completed dispatches for your account, so it's working a bit. I think it's possible that the 'plannedDispatch' wasn't coming through for last night because that was during the normal IOG cheap time anyway, so doesn't 'count' as a dispatch. 

This version has some fixes: https://github.com/Webreaper/SolisAgileManager/actions/runs/13324091881 and should show a log entry something like: 

```
[06:45:06.480--INF] Found 0 IOG Smart-Charge slots (out of a total of 0 planned and 4 completed dispatches)
```

I've refactored the code so the dispatches are queried every 5 minutes now, to avoid the scenario where:
1. Planned dispatch comes through at 16:07
2. We start charging the inveter for the 16:00-16:30 slot
3. At 16:15 you unplug your car and pop out for some shopping
4. We continue charging the inverter at what is now the peak rate, from 16:15 to 16:30

What should now happen is that you unplug the car at 16:15, and within 5 minutes we'll cancel charging. 

I don't know if this is how it will work, without seeing a 'live one' but that's how it _should_ work, I think. If you can install the build above, and try calling for your car to charge outside the IOG overnight cheap rate, we can see if it works.

dqj999, 2025-02-14:
That’s great. Thanks so much for doing this work. I’ll update to the new code in the next couple of hours and see if I can persuade Octopus that my car desperately needs charging to 100% this morning!

dqj999, 2025-02-14:
Just saw this in last night's log:

03:30:01.957--ERR] Unexpected exception during IOG Dispatch Query
System.Text.Json.JsonException: The JSON value could not be converted to SolisManager.APIWrappers.OctopusAPI+KrakenPlannedDispatch. Path: $.data.plannedDispatches[0].delta | LineNumber: 0 | BytePositionInLine: 116.
 ---> System.InvalidOperationException: Cannot get the value of a token type 'String' as a number.
   at System.Text.Json.ThrowHelper.ThrowInvalidOperationException_ExpectedNumber(JsonTokenType tokenType)
   at System.Text.Json.Utf8JsonReader.TryGetDouble(Double& value)
   at System.Text.Json.Utf8JsonReader.GetDouble()
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonConverter`1.TryReadAsObject(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, Object& value)
   at System.Text.Json.Serialization.Converters.LargeObjectWithParameterizedConstructorConverter`1.ReadAndCacheConstructorArgument(ReadStack& state, Utf8JsonReader& reader, JsonParameterInfo jsonParameterInfo)
   at System.Text.Json.Serialization.Converters.ObjectWithParameterizedConstructorConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonCollectionConverter`2.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, TCollection& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.Converters.SmallObjectWithParameterizedConstructorConverter`5.ReadAndCacheConstructorArgument(ReadStack& state, Utf8JsonReader& reader, JsonParameterInfo jsonParameterInfo)
   at System.Text.Json.Serialization.Converters.ObjectWithParameterizedConstructorConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.Converters.SmallObjectWithParameterizedConstructorConverter`5.ReadAndCacheConstructorArgument(ReadStack& state, Utf8JsonReader& reader, JsonParameterInfo jsonParameterInfo)
   at System.Text.Json.Serialization.Converters.ObjectWithParameterizedConstructorConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonConverter`1.ReadCore(Utf8JsonReader& reader, T& value, JsonSerializerOptions options, ReadStack& state)
   --- End of inner exception stack trace ---
   at System.Text.Json.ThrowHelper.ReThrowWithPath(ReadStack& state, Utf8JsonReader& reader, Exception ex)
   at System.Text.Json.Serialization.JsonConverter`1.ReadCore(Utf8JsonReader& reader, T& value, JsonSerializerOptions options, ReadStack& state)
   at System.Text.Json.JsonSerializer.ReadFromSpan[TValue](ReadOnlySpan`1 json, JsonTypeInfo`1 jsonTypeInfo)
   at SolisManager.APIWrappers.OctopusAPI.GetIOGSmartChargeTimes(String apiKey, String accountNumber) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/APIWrappers/OctopusAPI.cs:line 182
   at SolisManager.Services.InverterManager.ApplyIOGDispatches(IEnumerable`1 slots) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/Services/InverterManager.cs:line 825
[03:30:02.637--INF] Skipping charge request (Inverter state matches: 18, 0, 03:00-05:30, 00:00-00:00)
[03:30:02.712--INF] Refreshed state: SOC = 91%, Current PV = 0.000kW, House Load = 0.320kW, Forecast today: 3.51175kWh, tomorrow: 1.38145kWh
[03:30:03.184--ERR] Unexpected exception during IOG Dispatch Query
System.Text.Json.JsonException: The JSON value could not be converted to SolisManager.APIWrappers.OctopusAPI+KrakenPlannedDispatch. Path: $.data.plannedDispatches[0].delta | LineNumber: 0 | BytePositionInLine: 116.
 ---> System.InvalidOperationException: Cannot get the value of a token type 'String' as a number.
   at System.Text.Json.ThrowHelper.ThrowInvalidOperationException_ExpectedNumber(JsonTokenType tokenType)
   at System.Text.Json.Utf8JsonReader.TryGetDouble(Double& value)
   at System.Text.Json.Utf8JsonReader.GetDouble()
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonConverter`1.TryReadAsObject(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, Object& value)
[... cut ...]
   at SolisManager.APIWrappers.OctopusAPI.GetIOGSmartChargeTimes(String apiKey, String accountNumber) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/APIWrappers/OctopusAPI.cs:line 182
   at SolisManager.Services.InverterManager.ApplyIOGDispatches(IEnumerable`1 slots) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/Services/InverterManager.cs:line 825
[03:30:03.630--INF] Skipping charge request (Inverter state matches: 18, 0, 03:30-05:30, 00:00-00:00)
[03:30:03.631--INF] Enriching history with PV yield for 4 days
[03:30:03.758--INF] Retrieved 42 inverter stats for 14-Feb-2025
[03:35:01.110--ERR] Unexpected exception during IOG Dispatch Query
S

dqj999, 2025-02-15:
Just spotted the new option in Settings. I need to run down my EV battery so I can test the new Charge feature!

Webreaper, 2025-02-15:
There's a new chap in the FB group who asked for this too, so hopefully he can test too.

dqj999, 2025-02-15:
I’ve now got a day time charge session set up for 14:31 to 16:00 and from 16:30 to 16:46.

So in theory that means that I should get the reduced rate from 14:30 to16:00 and from 16:30 to 17:00. I don’t know what happens if the charge completes before some of the scheduled sessions.

dqj999, 2025-02-15:
[image]

Webreaper, 2025-02-15:
If SolisManager is working the tool tip for the slot should say it's an IOG dispatch. Is the inverter charging?

If the car finishes charging before the end of the scheduled charge, or the car is unplugged from the charger, the charge is cancelled. That's why I check the IOG state every 5 mins, so we don't carry on blissfully charging despite the price actually being peak.

dqj999, 2025-02-15:
it doesn't seem top have detected the charge:

[14:00:01.869--INF] Found 0 IOG Smart-Charge slots (out of a total of 0 planned and 0 completed dispatches)
[14:00:02.417--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[14:00:02.419--INF] Enriching history with PV yield for 4 days
[14:00:02.721--INF] Retrieved 168 inverter stats for 15-Feb-2025
[14:05:00.762--INF] Refreshed state: SOC = 50%, Current PV = 0.445kW, House Load = 0.371kW, Forecast today: 1.42790kWh, tomorrow: 3.43800kWh
[14:10:01.387--INF] Refreshed state: SOC = 50%, Current PV = 0.446kW, House Load = 0.345kW, Forecast today: 1.42790kWh, tomorrow: 3.43800kWh
[14:15:00.471--INF] Refreshed state: SOC = 50%, Current PV = 0.417kW, House Load = 0.417kW, Forecast today: 1.42790kWh, tomorrow: 3.43800kWh
[14:20:01.411--INF] Refreshed state: SOC = 50%, Current PV = 0.321kW, House Load = 0.321kW, Forecast today: 1.42790kWh, tomorrow: 3.43800kWh
[14:25:00.493--INF] Refreshed state: SOC = 50%, Current PV = 0.275kW, House Load = 0.201kW, Forecast today: 1.42790kWh, tomorrow: 3.43800kWh
[14:30:01.539--INF] Retrieved 4 rates from Octopus (16-Feb-2025 23:30 - 15-Feb-2025 23:30) for product E-1R-INTELLI-VAR-24-10-29-A
[14:30:02.431--INF] Found 0 IOG Smart-Charge slots (out of a total of 0 planned and 0 completed dispatches)
[14:30:03.154--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[14:30:03.156--INF] Enriching history with PV yield for 4 days

dqj999, 2025-02-15:
[image]

Webreaper, 2025-02-15:
Bit weird. Paul Johnson in the FB group said it worked for him?!

dqj999, 2025-02-15:
I'm wondering if Octopus is not giving me the cheap rate

[image]

dqj999, 2025-02-15:
I'm using Release 380.0. is that the latest one?

Webreaper, 2025-02-15:
If it has the IOG setting, yes.

But the IOG charging logic hasn't changed since yesterday morning.

dqj999, 2025-02-15:
One thing I noticed was that Ohme didn't want to schedule a smart charge session today. I thought that I had persuaded to do so, but perhaps I've forced it charge right away. If so the Ohme display is quite confusing.

Webreaper, 2025-02-15:
Could be!

dqj999, 2025-02-15:
It did find a smart charge slot eventually. It must be down to my scheduling or otherwise of the Smart-Charge.

[15:00:02.880--INF] Found 1 IOG Smart-Charge slots (out of a total of 1 planned and 0 completed dispatches)
[15:00:02.884--INF] Applying charge action to 3 slots for IOG Smart-Charge
[15:00:03.347--INF] Sending new charge instruction to Solis Inverter: 16, 0, 15:00-16:30, 00:00-00:00

The charge action on the inverter was carried out correctly/

BTW What's the name of the Facebook Group where this is being discussed.

Webreaper, 2025-02-15:
Yay!

It's the Solis Inverter group (where we first met). I'll try and tag you.

Nafeun1984, 2025-02-15:
Hey, I’m the new chap, just getting used to all of this. 
I’m running build 79 and it’s picking up my IOG charge slots nicely. The one thing it is not doing is recognising an expensive 30 minute window in the slots given to me. 
Tonight it was 1930-2000 and 2030 - 0400.
The app changed the inverter 30 min slots to charge BUT included the 2000 - 2030 slot.  I may just need to give it another 5 minutes to recognise it is not a cheap slot but I just paused that one manually.

Nafeun1984, 2025-02-15:
By the way. Thanks so much for the help so far. Helps my anxiety when the wife returns from work, plugs the car in and empties my house battery :)

Webreaper, 2025-02-15:
Next time can you give it 5-6 mins and see if it stops charging?

And could you send me the logs for today?

Nafeun1984, 2025-02-15:
Yes sure, how do I send you the logs?

Webreaper, 2025-02-15:
They're in the config folder, just add them as an attachment here (there's nothing sensitive in them).

Nafeun1984, 2025-02-15:
[SolisManager-20250215.log](https://github.com/user-attachments/files/18812948/SolisManager-20250215.log)

dqj999, 2025-02-15:
Nafeun, 

I'm trying this out too. What target charge and finish time do you normally set? 
I was wondering why it gave you such long cheap periods in the evening.

Nafeun1984, 2025-02-16:
I set the standard 2330 to 0530 30 minute slots up on the app. 
Octopus app set to 100% at 0530.
It then gives me a lot during the day, perhaps there is excess energy in the grid where I live?

Nafeun1984, 2025-02-16:
That’s with an ID4 with a 52kwh battery

Webreaper, 2025-02-16:
> [SolisManager-20250215.log](https://github.com/user-attachments/files/18812948/SolisManager-20250215.log)

I see there's an exception in this log, so there's an unexpected payload my code doesn't understand:
```
Uncaught Exception occurred: Invalid time pair 00:08-27:00
System.ArgumentException: Invalid time pair 00:08-27:00
   at SolisManager.APIWrappers.SolisAPI.ConvertToRealDates(String chargeTimePair) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/APIWrappers/SolisAPI.cs:line 126
   at SolisManager.APIWrappers.SolisAPI.InverterNeedsUpdating(Int32 chargePower, Int32 dischargePower, String chargeTimes, String dischargeTimes) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/APIWrappers/SolisAPI.cs:line 166
   at SolisManager.APIWrappers.SolisAPI.SetCharge(Nullable`1 chargeStart, Nullable`1 chargeEnd, Nullable`1 dischargeStart, Nullable`1 dischargeEnd, Boolean holdCharge, Boolean simulateOnly) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/APIWrappers/SolisAPI.cs:line 219
   at SolisManager.Services.InverterManager.ExecuteSlotChanges(IEnumerable`1 slots) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/Services/InverterManager.cs:line 363
   at SolisManager.Services.InverterManager.RecalculateSlotPlan(IEnumerable`1 sourceSlots) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/Services/InverterManager.cs:line 295
   at SolisManager.Services.InverterManager.SetManualOverrides(List`1 overrides) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/Services/InverterManager.cs:line 1055
   at SolisManager.Services.InverterManager.OverrideSlotAction(ChangeSlotActionRequest change) in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/Services/InverterManager.cs:line 969
   at SolisManager.APIWrappers.EndpointMapper.<>c.<<MapInverterAPI>b__1_4>d.MoveNext() in /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/APIWrappers/EndpointMapper.cs:line 57
--- End of stack trace from previous location ---
   at Microsoft.AspNetCore.Http.RequestDelegateFactory.ExecuteTaskResult[T](Task`1 task, HttpContext httpContext)
   at Microsoft.AspNetCore.Http.RequestDelegateFactory.<>c__DisplayClass101_2.<<HandleRequestBodyAndCompileRequestDelegateForJson>b__2>d.MoveNext()
--- End of stack trace from previous location ---
   at Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddlewareImpl.<Invoke>g__Awaited|10_0(ExceptionHandlerMiddlewareImpl middleware, HttpContext context, Task task)
```
I've added extra logging in this build, could you run it and send me the output the next time that exception appears? https://github.com/Webreaper/SolisAgileManager/actions/runs/13354146207

Webreaper, 2025-02-16:
Figured it out. Turns out that when you query Solis's API for the current charge state, it sometimes returns something like this: 

`50, 0, 00:08-27:00, 00:00-00:00, 00:00-00:00, 00:00-00:00, 00:00-00:00, 00:00-00:00`

That 27 means 27 hours - or actually '3am'. Grrr

Have put in a workaround

Nafeun1984, 2025-02-16:
Ideal, we re getting there. I have plugged it in this afternoon and the charge hours don’t quite align? 
See images Mark :)

[image]
[image]

Webreaper, 2025-02-16:
Oh, looks like the 1630 dispatch is triggering charge at 1630-1700.

Can you leave it until about 1710 and see if it cancels charging at 1705?

Webreaper, 2025-02-16:
If you want to try this build it might get that right: https://github.com/Webreaper/SolisAgileManager/actions/runs/13356257180

Also, you'll see "IOG Dispatch" in the 'price type' column, which should make things clearer.

Nafeun1984, 2025-02-16:
Sorry just popping to the pub. If I have time when I get back, I ll try the new build

Webreaper, 2025-02-17:
Latest dev release ([here](https://github.com/Webreaper/SolisAgileManager/actions/runs/13366501554)) has a couple of fixes - it should show a little 'car charging' icon if you're in an IOG Dispatch, and also I've fixed the graphs (#108). 

If you guys can test today and see if it behaves as expected, and maybe confirm if the 'stops charging within a few minutes of unplugging' function works correctly, I'll do a proper release. :)

Nafeun1984, 2025-02-17:
Just installed 393 Mark, seems to be running lovely. Picked up my EV charge slots within 5 minutes. Will test further tonight and tomorrow. Thank you

Webreaper, 2025-02-17:
Fab, let me know if it doesn't behave!

Webreaper, 2025-02-17:
Oh and let me know if the charge icon changes to a charging car when an IOG charge is running. I can't test that!

Nafeun1984, 2025-02-17:
Perfect 

[image]

Webreaper, 2025-02-17:
Nice - any chance you could take a proper screenshot (without using your camera)? I can include it in the docs. 😁

Nafeun1984, 2025-02-17:
[image]

Nafeun1984, 2025-02-17:
that ok mate?

Webreaper, 2025-02-17:
That's lovely! Thank you!

martinpike69, 2025-02-18:
Hello, Martin Pike here. I'm in the process of installing SolisAgileManager, and I'm on IOG.

I've read the IOG thread above, and as it's mostly in words, not code, I think I understand it all.

Just a couple of observations if I may - and apologies if I've missed something.

First, if you get 5 minutes of a Smart Charge within a 30 minute period - so say 4pm to 4.05pm, then you also get 4.05pm to 4.30pm on the cheap rate.  As such, rather than cancelling the inverter charge asap, can you set it to cancel at the net 30 minute point? For my own setup, that 25 mins could get over 2kWh into my batteries.

Second, as well as inverter based charging, I will probably also set my immersion to heat (as well as the washing machine, tumble dryer, but I draw the line at ironing!).  As such, is there a way of alerting us to to the change in schedule?  Noting your previous comment, I am that one person who actually does have a (Tuya Wifi) CT Clamp around my EV charger supply, so I am actually alerted, but I'm thinking of others.

I hope this is ok to send. If I can be of any - mostly non techie - assistance, please let me know.

And I hugely appreciate the work you've put in here.This could easily save me a minimum of 20 minutes a day fiddling with my phone!

Kind regards,

Martin

Webreaper, 2025-02-18:
> First, if you get 5 minutes of a Smart Charge within a 30 minute period - so say 4pm to 4.05pm, then you also get 4.05pm to 4.30pm on the cheap rate. As such, rather than cancelling the inverter [...]

This is interesting. This is how I assumed it would work, because smart-meters onl;y give you 30 min slots, and can't record at a greater granularity than that. However, somebody elsewhere in the thread mentioned that if the car is unplugged the price can immediately revert - so I put in the 5-minute check for that. 

If you're absolutely sure that even 1 minute of EV charging within a 30-minute slot will get you the whole 30 minutes at the low price, then I can easily change the algorithm so that it won't cancel charging early in that scenario. But I need it to be really sure, otherwise I'll cost people money. :)

@dqj999  / @Nafeun1984  - can you confirm this is your understanding of how it works too?

> Second, as well as inverter based charging, I will probably also set my immersion to heat (as well as the washing machine, tumble dryer, but I draw the line at ironing!). As such, is there a way of alerting us to to the change in schedule? 

I'm really not keen to go down this route, as before long I'll have ended up building HomeAssistant. I think if you want that level of integrated coordination between all of your devices etc, you probably need to look at HA. 

That said, I'll have a think about it - if there is a simple way that I can provide some sort of notification (be it file-based or some sort of other notification thing) I might consider it. But this is quite far out of my own personal use-case (with no EV etc). I personally tend to work on the basis that if I want to heat the HW, or do some washing, that's why my battery is there - to avoid too much manual load-shifting where possible.....

martinpike69, 2025-02-18:
That makes perfect sense re alerts / messaging Mark.  I'm brand new to PI and Github, so have no idea if the platforms themselves have something built in.

I'll keep my Tuya WiFi CT Clamp in place, which then is able to switch on my immersion (and various other high load devices, including ASHP) automatically.

Re the 5 minute / 30 minute thing, I'm 95% sure it works like that, but I'd keep to the current logic until / if it is proven.  To be honest, I tend to check my Octopus bill, and if my average pence per kWh is less than 10p, I can't be bothered to compare the daily charts to the Ohme log.  But I will do so next time I receive a bill.

I'm just waiting for my Solis API to be sorted, everything else is in place.

Cheers,

Martin

Webreaper, 2025-02-18:
Sounds good. What I could do is add an option to say "stop charging as soon as the IOG Slot ends" and if it's unchecked it'll plough on and keep charging for the full 30 minutes. I want to try and avoid too many options as it makes it hard to use, debug and support though. Let's see what you find out.

martinpike69, 2025-02-18:
OK, here are some comparisons from my most recent bill.

Jan 4th. Started charging 1623, ended 2033.  Cheap rate slots from 1600 to 2030.
Jan 5th. Started charging 1505, ended 2221.  Cheap rate slots from 1500 to 2130.
Jan 6th. Started charging 1437, ended 2105.  Cheap rate slots from 1430 to 2130.
Jan 9th. Started charging 1839, ended 2112.  Cheap rate slots from 1830 to 2130.

So, we have some scenarios where the cheap rate ends before the charge does - apparently.

This *could* be explained by a few things. I almost always leave plenty of charging headroom in my car, to allow for tactical plugging in if I require an energy boost. However, this isn't always the case, and I've noticed that when the car reaches full (as alerted by the CT Clamp monitor), the Ohme charger tries to charge the car again, a bit like dribbling more petrol in your car. I can't see any 'mini' charges of 1-2 mins on the Ohme app log, but perhaps they send more granular data to Octopus, who then discard anything under 5 mins (or whatever).

Again, usually led by other energy requirements, I sometimes pause a charge via the Ohme app, followed a while later by physically unplugging. I don't keep records of this.

I don't think we're realistically going to get to the bottom of this, so I think you're right to consider asking each user whether they want to carry on the next half hourly end period, or stop asap after a charge.

I hope this helps.

Cheers,

Martin

Webreaper, 2025-02-18:
Interesting. 

It might be that whilst the smart meter can only record 30 minute slots, Octopus are directly controlling the EV charger, so can make adjustments. 

So for example, in a 30 min period:

* At minute zero it's 25p.
* At minute 5 they start you charging @ 7p
* At minute 15 you unplug the car

It's not hard to imagine that they bill you for your entire energy usage for the 30 minute period, but then they pro rate 30% of it to 7p to cover the 10 minutes while you were charging. 

So technically if you thrashed the grid charging your inverter for the last 15 minutes you'd get it at a bit cheaper than 25p because of the pro rata calc, but on the whole it would work out fine for the customer and octopus. 

What we really need is somebody with a Home Mini and an EV on IOG so they can watch it charge at the cheap rate, then unplug their EV and see if the rate changes...

martinpike69, 2025-02-18:
I don't have a Home Mini, but if I press the right buttons my actual smart meter tells me what rate I'm paying - live...

[85 more replies]

closed 2025-02-27 by Webreaper