Webreaper/SolisAgileManager #111: Zero's at 5 past the hour

By-tor, 2025-02-15:
Is there any possibility that the Solis Manager app could be causing this issue with the Solis Cloud app returning all zeros (presumably can't access my inverter) at 5 minutes past every hour?

[image]

[image]

Webreaper, 2025-02-15:
I supposed it's possible that the app is triggering some sort of rate limiting? But I don't see this.

Which version are you running?

Webreaper, 2025-02-15:
Thinking abut it, the app doesn't carry out any activities at 5 past the
hour.

Perhaps try stopping it running, and see if the 5-past failures stop.

Webreaper, 2025-02-15:
When did you first start running the app?

And have you done any updates?

The app does make an inverter API call every 5 minutes to get the SOC and other stats, but that shouldn't cause any issues.

By-tor, 2025-02-15:
At 5 past my Solis Cloud app flips to this display too instead of the usual. I have logged a service ticket with Solis.

[image]

Webreaper, 2025-02-15:
I get that when the Solis API is having issues....

By-tor, 2025-02-15:
It's been running since 08:30 on 13th February, I haven't made any updates since the original download of the Windows zip file on the 7th February.

I have another inverter (different site) running fine on Solis Cloud app without issues.

I will see what Solis come up with the try stopping your app to see what happens

Webreaper, 2025-02-15:
If you stop the app and the problem goes away, please send me your logs.

In fact, can you add the logs here anyway so I can check nothing is untoward?

Do you run any other stuff that uses the Solis API, like HA etc?

By-tor, 2025-02-15:
we're just coming up to the hour and Solis seem to have put right again (for the second time since I logged the ticket) so I will stop SAM and see what happens.

Not running anything else that uses the Solis API

I'll grab the logs now

By-tor, 2025-02-15:
[SolisManager-20250215.log](https://github.com/user-attachments/files/18811718/SolisManager-20250215.log)

[SolisManager-20250214.log](https://github.com/user-attachments/files/18811720/SolisManager-20250214.log)

[SolisManager-20250213.log](https://github.com/user-attachments/files/18811721/SolisManager-20250213.log)

Webreaper, 2025-02-15:
No errors or anything else untoward in the logs.

By-tor, 2025-02-15:
It's 16:15 now and the solis cloud display is fine without zero's a 5 past.
It was actually zero's at 5 past the hour AND half hour.
I guess your logs won't show errors if it can get access but the Solis Cloud app can't for some reason.

[image]

Webreaper, 2025-02-15:
@dqj999 - have you seen this issue occurring, at all...?

By-tor, 2025-02-15:
I've set it running again now to see what happens at 16:35

Webreaper, 2025-02-15:
Could you try running the latest dev build here? I've changed a lot of the background updates.

https://github.com/Webreaper/SolisAgileManager/actions/runs/13343143724

If this is contention with the Data logger, it might be solved by changes I'm considering to add 'jitter' to the API calls. So when each API call is made, it'll be done at a random time within a 10s window, so not all the calls happen at the same time. Primarily to avoid DDoSing Solcast and Octopus, but I might add it to the Solis calls too.

It's be interesting to see what Solis say.

Webreaper, 2025-02-15:
One other question - do you have the setting enabled to update the inverter time each day?

By-tor, 2025-02-15:
Yes - Auto Update Inverter Time is enabled

By-tor, 2025-02-15:
Surprisingly, since I stopped the app at 15:55 and re-started it at 16:15 the have been no more zero's !!

I will leave it running for an hour or two and see what happens, then I will update it to SolisManager-windows-1.0.380-dev.zip

By-tor, 2025-02-15:
No response from Solis yet although I know they have been looking at the issue - they put a response almost immediately then deleted it just as quickly

Webreaper, 2025-02-15:
> Surprisingly, since I stopped the app at 15:55 and re-started it at 16:15 the have been no more zero's !!
> 
> I will leave it running for an hour or two and see what happens, then I will update it to SolisManager-windows-1.0.380-dev.zip
> 

That **is** weird.

By-tor, 2025-02-15:
Spoke too soon - just threw a wobbler at 17:05 !!

I was going to update to SolisManager-windows-1.0.380-dev.zip but I have found I am at the edge of my knowledge of github !!
How do I get it?

Webreaper, 2025-02-15:
Go to the link I posted and click the "Artifacts" link (it's a '5') in the top right corner.

By-tor, 2025-02-15:
Apologies - wasn't rocket science - I wasn't logged in to git hub on the virtual machine so it didn't show the download icon

By-tor, 2025-02-15:
Since 17:45 when applying the update and running on 1.0.380 there have been no more zero's in the data.
I will leave this running and see how it goes.
Solis still have not responded.

Webreaper, 2025-02-15:
Good to hear.

By-tor, 2025-02-15:
Spoke too soon  - again 

I notice that charging strategy graph, charging history graph and tarrif comparison pages are not working in this version,  are you aware of that?

Webreaper, 2025-02-15:
Yeah, something weird with the graphs I need to fix. Shift-F5 makes them work again.

Webreaper, 2025-02-15:
For the 5 minute thing wait until Monday and I'll get the jitter fix in to see if it stops colliding with Soliscloud hitting the API (if that's what it is).

Webreaper, 2025-02-16:
@By-tor could you please try this build? It adds a random delay of 50ms - 3000ms before each Solis API call, which may be enough to solve this issue.

By-tor, 2025-02-16:
Won't be able to try this till tonight sorry, out for the afternoon.

Not sure if it's important but Solis's cloud server time appears to be 4 minutes ahead, i.e. it reports 12:10 datapoints at 12:06

Webreaper, 2025-02-16:
Only Solis could run something so time sensitive without accurate time!

Just try when you get the chance. If it solves it, I'll implement the feature properly (I just hacked it up to check if it solves your issue).

By-tor, 2025-02-16:
Downloaded v1.0.381.0 and running now - we shall see what happens

By-tor, 2025-02-17:
So v1.0.381.0 ran all night with the issue just the same, zero data in solis cloud app every 30 minutes.

[image]

Then this morning I finally got a response from Solis

[image]

And lo and behold 

[image]

My apologies for wasting your time with what was obviously a Solis Cloud App issue but it was difficult to tell where the issue lay.
It was I suspect Solis's app which had no routine to discard failed data requests and retry.
Interestingly, their server time seems much closer to real time since their update, only seconds difference now.

Should I return to v1.0.349.0  now?

closed 2025-02-17 by Webreaper