Webreaper/SolisAgileManager #113: Dark Mode - Initial icon wrong for Charging Plan and History

dqj999, 2025-02-17:
In dark mode when you first select a view eg Home or History, the Icon that says whether you are in Table Mode or Graph mode is rendered with a Black Foreground and a Dark Grey Background rather than a Black foreground on a White background. 

Using Chrome on my iPad if you reselect the mode it then corrects itself to Black on White (not on my PC though).

[image]

This also happens on the history view in dark mode

[image]

Webreaper, 2025-02-17:
It's actually working correctly - but the colours are a poor choice for dark mode. I'll fix.

dqj999, 2025-02-17:
Photo of iPad view after reselecting the view

[image]

Webreaper, 2025-02-17:
I think there's something odd about how iPads render the `:hover` CSS class. Anyway, I've made it a bit better.

closed 2025-02-17 by Webreaper

Webreaper, 2025-02-17:
Just checked. It's just showing the hover state because on a touch screen the focus remains there until you tap away. If you tap on anything else the gray background disappears. So it's as designed.

dqj999, 2025-02-17:
Yep - all fixed!