Webreaper/SolisAgileManager #118: Calculate solcast dampening % using actual PV generation data

tabannis, 2025-02-21:
Suggestion: Using data over time from the inverter daily PV generation stats, calculate the solcast dampening percentage dynamically as well as having a constant option.

Webreaper, 2025-02-21:
Already logged (and I've been investigating this now there's some data). https://github.com/Webreaper/SolisAgileManager/issues/37

dqj999, 2025-02-22:
One thing I would suggest is that this needs to be done over each 30 min
period because in that way the adjusted forecasts will take into account
shading effects that occur at particular times of the day.

If you use something like a 2 week rolling average that can take account of
the gradually changing seasonal effects as well.

Derek

Webreaper, 2025-02-22:
Yep, that's the general idea.

dqj999, 2025-03-02:
I've created an Excel Spreadsheet that you can use to analyse your results from the SolisManager .csv file.

It's in this :

[GitHub project  - SolCastAnalysis](https://github.com/dqj999/SolCastAnalysis)

Derek

Webreaper, 2025-03-02:
Handy, thanks!