Webreaper/SolisAgileManager #119: Support new Solis API v2 for S6 inverters

Webreaper, 2025-02-22:
Add detection and support for the new v2.0 Solis inverter API (forked from #107).

[SolisCloud.Device.Control.API.V2.0.new.pdf](https://github.com/user-attachments/files/18921370/SolisCloud.Device.Control.API.V2.0.new.pdf)

Webreaper, 2025-02-22:
@martinpike69 - could you send me a screenshot of the inverter control screen where you set up the charge slots please?

martinpike69, 2025-02-22:
Yes, my old S5 (I think) / 6kw inverter used to charge and discharge at 6kw for 15 minutes (to the second), and then de-rate to 3.6kw.  Which, as far as I'm concerned, is a 3.6kw inverter that bursts to 6kw for 15 minutes, not a 6kw inverter.

Seeing as I have 9kw'ish of solar panels, and 9 x 2.4kw Pylontech US2000C batteries, the inverter was a constant bottleneck.

Fortunately, my installer was superb about it, and was annoyed as I was that his design - which was perfectly well balanced based on the spec sheet (which didn't mention the de-rating at all) - wasn't performing as it should.

They therefore - rather reluctantly - replaced the device at no cost to me.

Re charging and discharging, it has 6 slots.  I've attached screenshots of slots 1 and 2...

[image]

Webreaper, 2025-02-22:
Weird, we have the same S5 6kw, and it happily sustains 6kw,and sometimes 7-8kw for quite some time.

martinpike69, 2025-02-22:
I'm assuming that for charging it's the target / when to stop, and for discharging it's the minimum...?

martinpike69, 2025-02-22:
I only use 100% SOC for Charging, and 10% SOC for discharging (even though my US2000C's are meant to operate down to 5%).

Webreaper, 2025-02-22:
More research: 
* Info here on how to check the firmware version: https://github.com/hultenvp/solis-sensor/blob/master/custom_components/solis/soliscloud_api.py#L447

Webreaper, 2025-02-22:
If you get 5 mins, Martin, can you check your firmware version, following these steps?

https://usservice.solisinverters.com/support/solutions/articles/73000558553-how-to-determine-the-firmware-version-of-your-solis-inverter

Webreaper, 2025-02-22:
Another reference - looks better. https://github.com/hultenvp/solis-sensor

martinpike69, 2025-02-22:
I'm working in the garden today, but I'm more than happy to take a break from that to help you to help me!

The S6 doesn't have a screen as such, control is completely via the app, so I can't follow those instructions - although I remember them from the previous model.

The closest I can find in the app is 'Inverter Version', which then has 1D0010-62000d after it.

Would that be what you're looking for?

Webreaper, 2025-02-22:
That's perfect.

I've finished my gardening for today. 😁👌

fboundy, 2025-02-23:
Hi guys. Looks like you’ve found what you were looking for in my HomeAssistant code.

I’ve written something very similar in Home Assistant here: https://github.com/fboundy/pv_opt and I’m currently converting it to a full blown Home Assistant integration.

Interestingly your optimisation algorithm appears to use very similar logic to mine. 

Have you considered adding direct control of the inverter via TCP/Modbus? It’s pretty straightforward in Python - no idea how hard it would be in C.

Webreaper, 2025-02-23:
Most of these algorithms are "charge when it's cheapest, plus some other tweaks". 😁

Haven't looked at Modbus - I'm more interested in the vanilla Soliscloud connection (my understanding is that if you use Modbus it prevents you using Soliscloud).

I think I'm understanding your python code - will have a go at an implementation on Monday.

Cheers
Mark

fboundy, 2025-02-23:
I think for most folk the cloud API version is fine - local Modbus does seem to clash with the API unless you add extra hardware.

Let me know if anything isn’t clear in the Python code

Webreaper, 2025-02-24:
@martinpike69 Just successfully test-charged your inverter for 5 minutes.... :)

```
[12:50:12.241--INF] Refreshed state: SOC = 18%, Current PV = 1.625kW, House Load = 1.072kW, Forecast today: 12.04430kWh, tomorrow: 19.53935kWh
[12:50:12.243--INF] Updating inverter time to avoid drift...
[12:50:12.243--INF] Simulated inverter control request: { inverterSn = 1031150234140015, cid = 56, value = 2025-02-24 12:50:12 }
[12:50:20.826--INF] Saving config to server...
[12:50:20.953--INF] Retrieved 12 rates from Octopus (25-Feb-2025 22:00 - 24-Feb-2025 13:00) for product E-1R-COSY-22-12-08-J
[12:50:25.597--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[12:50:25.598--INF] Enriching history with PV yield for 2 days
[12:50:25.950--INF] Retrieved 153 inverter stats for 24-Feb-2025
[12:50:25.978--INF] Starting test charge for 5 minutes
[12:50:30.791--INF] Setting charge Amps to 75A for slot 1
[12:50:35.449--INF] Setting charge Amps to 100% for slot 1
[12:50:38.717--INF] Setting charge time to 12:50-12:55 for slot 1
[12:55:00.442--INF] Refreshed state: SOC = 22%, Current PV = 1.651kW, House Load = 2.876kW, Forecast today: 12.04430kWh, tomorrow: 19.53935kWh
```
Note the SOC goes from 18% => 22%. :)

This is good progress - I just need to do more testing and tidy the code up and I should have a working version for you to beta-test soon....

@fboundy  - you can ignore the above post, I think I've got it now. Thanks for your assistance with this!

martinpike69, 2025-02-24:
Ah, that will explain why my SolisCloud has been playing silly buggers over the past 30 mins or so?!

:-)

Webreaper, 2025-02-24:
Oh, sorry, yes.... 😁

Webreaper, 2025-02-24:
BTW, I think you had slot 1 set to charge from 21:00 - 22:30, @ 75% or something - you'll need to reset it.

fboundy, 2025-02-24:
Yep - looks right to me.

fboundy, 2025-02-24:
Out of interest, where do you get your forecast PV generation from?

Also, for Agile, you might want to use my agilepredict.com API to get predicted prices

Webreaper, 2025-02-24:
PV forecast comes from Solcast. There's basically two providers:
 
* https://forecast.solar/
* https://solcast.com/

I use solcast because it's pretty easy, but also gives 30-minute predictions for free, for up to 2 strings (with a rooftop hobbyist account). Forecast.Solar isn't so good for free.

Thanks for the agilepredict suggestion, but I'm not planning to integrate agile prediction into the app right now (or, probably, ever 😆 )

Webreaper, 2025-02-24:
@martinpike69 if you want to give this version a go, I think it should work, mostly?

https://github.com/Webreaper/SolisAgileManager/actions/runs/13500336045

Can you try: 
1. Test charge from the settings screen
2. Test discharge using the manual overrides
3. If those work, let it run for a bit and see if it works and does something sensible.

martinpike69, 2025-02-24:
Sure. When I updated last time I lost all my login data to Solcast / Solis / Octopus.

Am I best to switch the PI off and on again, and download the new version?  Then extract it to which folder please?  And to not overwrite files - as I assume this is how I lost my logins last time?

Thanks,

Martin

Webreaper, 2025-02-24:
Just extract to copy over the old files. If you're worried about losing
your login data, copy the config folder somewhere else, and restore it
after the copy is complete.

You should be able to use the "shutdown" option on the Tools page, you
shouldn't need to power cycle the Pi.

martinpike69, 2025-02-24:
I still don't have the Shutdown option on the Tools page?

Just Charge / Discharge and Dump and Recharge?

Webreaper, 2025-02-24:
Did you ever upgrade to the latest dev build? If not, you won't have it....

Upgrade today, and use the shutdown commands that Derek posted in the IOG thread, and you'll have it after that.

martinpike69, 2025-02-24:
I'm clearly doing something wrong. I have a folder called SolisAgileManager, and under that another one called SolisManager-raspi.  I then have the new version - SolisManager-raspi-1.0.412-ev.zip.zip -  in my Downloads folder. Which of the Solis folders am I extracting the downloaded file to please?

Webreaper, 2025-02-24:
I suspect you have:

  `./SolisManager/SolisManager-raspi`

So you need to:

* Extract the `SolisManager-raspi-1.0.412-ev.zip.zip` into a temporary folder
* Then copy the contents of that folder (which should be around 100 files etc) into the `./SolisManager/SolisManager-raspi` folder.

martinpike69, 2025-02-24:
I now have the Restart option.

5 minute test charge from Settings screen has worked.

Manual Charge from Home Screen seems to work

Manual Discharge **doesn't** seem to work?  Is this meant to in effect dump as much battery as possible, exporting anything the house isn't using?

Webreaper, 2025-02-24:
> I now have the Restart option.
> 5 minute test charge from Settings screen has worked.
> Manual Charge from Home Screen seems to work

Hoorah!

> Manual Discharge doesn't seem to work? Is this meant to in effect dump as much battery as possible, exporting anything the house isn't using?

Yes, that's what's supposed to happen. It might be that I haven't got the new API quite right for the discharge. I'll take a look at that tonight, but the rest sounds promising. :)

martinpike69, 2025-02-24:
Have you got any plans to amend the Price (p/kWh) on the Home Screen to reflect 7p / Off Peak as and when a cheap daytime slot is scheduled under IOG?  I do of course appreciate that this is an Agile focussed system...

Webreaper, 2025-02-24:
Can you send me a screenshot of what you mean? 

The off-peak (overnight) price for IOG should show correctly, because that's returned from the tariff data. 

For the IOG smart-charge dispatches I don't get any information about the price, so can't display anything there unless I hard-code it....

martinpike69, 2025-02-24:
Ah, I understand. I assumed that a smart charge dispatch would also come with some sort of 'off peak' indicator, even if not a price.  No problem.  If would probably set some minds at rest if the 'Peak' word was hard coded to change to Off Peak during any smart charge periods though?  Completely up to you.

Webreaper, 2025-02-24:
Yes, you're right, the price-type should switch to 'off-peak' when an IOG dispatch is in-flight (I thought I'd done that, actually).

Webreaper, 2025-02-24:
Just checked the code - if the car icon is displaying, the price indicator should change from 'Peak' to 'EV Charge'. Is that not happening?

Webreaper, 2025-02-24:
> Manual Discharge doesn't seem to work? Is this meant to in effect dump as much battery as possible, exporting anything the house isn't using?

By the way, when you tried this, what was your SOC? It won't let you discharge below a certain level.

Webreaper, 2025-02-24:
@fboundy I can't spot anything wrong with my Discharge code, it looks like it should work. Is there anything special that needs to be done for the Discharge control - like set the amps to negative or anything odd like that?

Webreaper, 2025-02-24:
Okay. Might try some test discharges tomorrow, if that's okay? (only a couple of minutes, once I get it to work).

fboundy, 2025-02-24:
Not that I recall. Are you setting the Volts for each slot (in the new FW)

Webreaper, 2025-02-24:
> Not that I recall. Are you setting the Volts for each slot (in the new FW)

I'm setting the Amps. Or is there a Volts setting separate from the amps?

Webreaper, 2025-02-24:
Okay, I'm still struggling to figure out what's wrong with the discharge settings. 

Currently I do this when charging:
- Set the charge SOC % to 100%
- Set the discharge SOC % to 15%
- Set charge time to (say) 21:30-22:00
- Set charge amps to 75A
- Set discharge time to 00:00-00:00
- Set discharge amps to 0

And I do this for discharging:
- Set the charge SOC % to 100%
- Set the discharge SOC % to 15%
- Set charge time to 00:00-00:00
- Set charge amps to 0
- Set Discharge time to (say) 21:30-22:00
- Set discharge amps to 75A

I wonder if the problem is that I'm setting the charge SOC % to 100%, which is setting a charge target - is this conflicting with the discharge? Should I set the Charge SOC to 15% or clear it somehow?

fboundy, 2025-02-24:
There is in the new Firmware but there is no documentation as to what it does! My recollection is that the app sets it low (49.5v) on charge but I can’t remember if it is high on discharge. Will check when I have time

Webreaper, 2025-02-25:
Okay, looks like I was right. If you set a discharge time, and set the discharge amps to (say) 50A, and you have a ChargeSOC of 100%, then nothing happens. 

If I change it so that the charge time and discharge amps are set, and then change the ChargeSOC to 15%, it discharges.

Thanks to @martinpike69 for letting me test with his inverter (I've stopped buggering about now). If you want to try a build which should now be fully functional for charge, discharge, and 'do nothing', try this: https://github.com/Webreaper/SolisAgileManager/actions/runs/13515324000
The only function I haven't explicitly tested is 'hold'.

martinpike69, 2025-02-25:
I've just managed to install - at the second attempt, I forgot to reboot my PI first time - the latest dev version.

I've also reinstated Solcast, which seems to be pulling data in.

I'll be plugging in the car later today, when I'll have a look for a car icon or similar on the Home screen.

Thank you so much!

Webreaper, 2025-02-25:
Great. Let me know if discharge works for you too - I did try it, but it's easier to debug if you can go to the inverter and see what it's doing live on the screen. :)

martinpike69, 2025-02-25:
Not sure my system is working.

I plugged my car in at 0930, and it's charging.

We're now 0938 and my house battery isn't charging.  I thought the calls to Octopus were every 5 mins?

Have I misunderstood something?  If I have, please explain.  If I haven't, should I send you the log file for today?

Thanks,

Martin

martinpike69, 2025-02-25:
Still not charging at 0945, so I pressed the Charge icon under Manual Actions.  That doesn't seem to have worked either.

martinpike69, 2025-02-25:
Sorry, forget the above two comments - user error my end!!

Webreaper, 2025-02-25:
So it's all working? 😁

martinpike69, 2025-02-25:
Actually, no, sorry!

Re-plugged in at 0950, got given immediate charge slot. Now 10am, and house battery not charging. I can see that an update re rates has been received from Octopus.

Log file attached.

martinpike69, 2025-02-25:
I tried to attach it...

[SolisManager-20250225.log](https://github.com/user-attachments/files/18961998/SolisManager-20250225.log)

martinpike69, 2025-02-25:
And Manual Charge and Discharge don't seem to be working either.  Sorry!

Webreaper, 2025-02-25:
Debugging and testing now - your inverter might do some stuff. :)

martinpike69, 2025-02-25:
If you can leave it until 11am if at all possible please?  I'm currently recharging the house batteries...

Webreaper, 2025-02-25:
Sorry, missed this message - I ran a test charge which seemed to work, but may be because you were already charging. I'll leave it for a bit.

fboundy, 2025-02-25:
When changing leave the discharge setting off and vice versa. To do this I set start and end times the same.

Webreaper, 2025-02-25:
> When changing leave the discharge setting off and vice versa. To do this I set start and end times the same.
> […](#)

Yeah, I'm doing that already - if I'm charging I set the discharge start/end times as 00:00, and for a discharge I set the charge start/end times to be 00:00. But it still seems that if you want to discharge, and you have a 00:00-00:00 charge time set, but the Charge SOC is 100%, the inverter won't discharge. If I set the charge SOC to 15%, it then discharges...

Webreaper, 2025-02-25:
@martinpike69 I just did some testing, and from what I can see it's working. Here's the log:

```
[11:40:51.273--INF] ===========================================================
[11:40:51.294--INF] Application started. Build version v1.0.0.0 Logs being written to ./config
[11:40:51.468--INF] Loaded cached Solcast data from ./config/Solcast-cache.json
[11:40:52.361--INF] Now listening on: http://localhost:5158
[11:40:52.362--INF] Application started. Press Ctrl+C to shut down.
[11:40:52.362--INF] Hosting environment: Development
[11:40:52.362--INF] Content root path: /Users/markotway/LocalCloud/Development/SolisAgileManager/SolisAgileManager/SolisManager
[11:40:53.105--INF] Retrieved 12 rates from Octopus (26-Feb-2025 22:00 - 25-Feb-2025 13:00) for product E-1R-COSY-22-12-08-J
[11:40:53.109--INF] 73 new Octopus rates available to 27-Feb-2025 00:00 (cheapest: 13.24239p/kWh, peak: 40.504905p/kWh)
[11:40:53.120--INF] Loaded 1234 entries from execution history file SolisManagerExecutionHistory.csv
[11:40:53.148--WRN] SOC is zero, so skipping charge if low battery due to bad inverter state data
[11:40:53.149--WRN] SOC is zero, so skipping maintain charge rule due to bad inverter state data
[11:40:57.209--INF] Detected new firmware version: 43605 (AA55)
[11:40:59.166--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[11:40:59.185--INF] Enriching history with PV yield for 2 days
[11:40:59.550--INF] Retrieved 135 inverter stats for 25-Feb-2025
[11:40:59.994--INF] Retrieved 264 inverter stats for 13-Feb-2025
[11:41:00.163--INF] Refreshed state: SOC = 42%, Current PV = 3.347kW, House Load = 5.968kW, Forecast today: 12.10745kWh, tomorrow: 4.93330kWh
[11:41:00.165--INF] Updating inverter time to avoid drift...
[11:41:00.167--INF] Simulated inverter control request: { inverterSn = 1031150234140015, cid = 56, value = 2025-02-25 11:41:00 }
[11:41:19.973--INF] Saving config to server...
[11:41:20.083--INF] Retrieved 12 rates from Octopus (26-Feb-2025 22:00 - 25-Feb-2025 13:00) for product E-1R-COSY-22-12-08-J
[11:41:21.694--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[11:41:21.699--INF] Enriching history with PV yield for 2 days
[11:41:22.065--INF] Retrieved 135 inverter stats for 25-Feb-2025
[11:41:32.337--INF] Updating slot action for 02/25/2025 11:30:00 to Charge...
[11:41:32.341--INF] Set override: 11:30 - Charge
[11:41:34.082--INF] Sending new charge instruction to Solis Inverter: 75, 0, 11:30-12:00, 00:00-00:00, SOC: 100%, D-SOC: 15%
[11:45:00.708--INF] Refreshed state: SOC = 43%, Current PV = 2.453kW, House Load = 5.872kW, Forecast today: 12.10745kWh, tomorrow: 4.93330kWh
[11:50:00.578--INF] Refreshed state: SOC = 45%, Current PV = 2.389kW, House Load = 7.177kW, Forecast today: 12.10745kWh, tomorrow: 4.93330kWh
[11:50:05.123--INF] Updating slot action for 02/25/2025 11:30:00 to DoNothing...
[11:50:05.123--INF] Cleared override: 11:30 - DoNothing
[11:50:10.429--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00, SOC: 100%, D-SOC: 15%
[11:55:00.665--INF] Refreshed state: SOC = 43%, Current PV = 2.296kW, House Load = 9.408kW, Forecast today: 12.10745kWh, tomorrow: 4.93330kWh
```
If you look at the `Refreshed State` log entries for `11:41:00`, `11:45:00` and `11:50:00`, you can see that the battery was indeed charging - it went from 42% => 45%. This despite the fact that your house load is a whopping 9.4kW, of which only around 2-3kW is being supplied from PV. 

So it looks like charging is working. I haven't tried discharge, because a) that's less critical, and b) it'll be difficult to tell if it's working because the house load and PV are high at the moment. We can try that another time.

So when you said "Manual Charge and Discharge don't seem to be working" we need to understand what's happening there. 

Let's focus on that for the moment and ignore the IOG smart-charge - we've already seen from Derek's experiences that sometimes even though your EV is charging, Octopus don't send a smart-charge notification so SolisManager may not get notified and won't attempt to charge the inverter/battery. In the log you sent 2 hours ago there certainly weren't any IOG dispatches picked up by SolisManager, so I don't think they were sent from Octopus. If you don't see a car icon instead of the lightning icon, then the IOG notification hasn't come through from Octopus.

Once we've established that the changes for the new firmware are working as expected (and from my end, that looks to be the case) then you can run for a few days and see if the IOG Dispatches start showing up.

martinpike69, 2025-02-25:
Now heading out, but I did a charge via SolisCloud when this system didn't work.
We often have a high load.  ASHP, immersion, EV charging, cooking etc.
Have you seen the new email from Ohme?  Implementing an end time of 4am to 11am, like all other IOG integrations.

Thanks and regards,
Martin

Webreaper, 2025-02-25:
> Now heading out, but I did a charge via SolisCloud when this system didn't work.

Don't understand this comment. 

> We often have a high load.  ASHP, immersion, EV charging, cooking etc.

Yeah, us too. That's fine, it just means dicharge isn't always obvious since the battery is probably being chewed up anyway. But like I say, we can ignore testing manual discharge as it's an edge case and isn't really important unless Agile price go negative.... 😄 

>  Have you seen the new email from Ohme?  Implementing an end time of 4am to 11am, like all other IOG integrations.

Email? Not sure what you mean. I wouldn't see any emails from your Ohme charger. The only notifications that I (or SolisManager) will see are those provided by the Octopus API...

Let's catch up later...

martinpike69, 2025-02-25:
I just meant that when pressing Charge under Manual Actions didn't work, I did it via soliscloud.
M

Yahoo Mail – Email Simplified

[44 more replies]

closed 2025-02-27 by Webreaper