Webreaper/SolisAgileManager #124: Docker on RasPi not working

tabannis, 2025-02-28:
I know this is already known about, but the side-issue where this was previously discussed has been closed.

The docker container for Raspberry Pi releases isn't working. For me it generates the following error in docker on run...

`exec /app/SolisManager: no such file or directory`

My system

Raspberry Pi 5 running LibreElec, Docker 27.5.1, using the following docker compose config:

```
solismanager:
        container_name: solismanager
        image: webreaper/solisagilemanager:latest
        ports:
            - '5168:5168'
        restart: unless-stopped
        volumes:
            - /storage/system/solismanagerdocker:/appdata
```

Webreaper, 2025-02-28:
Yep, I need to fix this, but unsure how at present!

Is your Pi an Arm64?

tabannis, 2025-02-28:
Yup, it is indeed Arm64

Webreaper, 2025-02-28:
Thanks. Yeah, may be a bad base image.

dqj999, 2025-03-02:
This isn't the cause of the problem, but I thought that it was port 5169 that was used rather than 5168. That's certainly what my working RasPi system is using. 

There do seem to be other solar management systems that work with Docker on RasPis but I've don't have enough knowledge to figure out what needs to be done.

tabannis, 2025-03-02:
I changed the port number to 5168 in the yaml so it wouldn't conflict with my (Working) raspi installation on 5169 👍

Webreaper, 2025-03-03:
You need to change the port on one side, not the other. The port internal to the container is always 5169; if you want to expose an alternative port from the container then the port setting should be:

`5168:5169`

tabannis, 2025-03-03:
Ah, that's good to know. Thanks, I'll update my configuration,👍

baruba-cons, 2025-03-04:
Shouldn't `platforms: ${{ matrix.platform }}` be `platforms: ${{ matrix.docker-platforms }}`?

https://github.com/Webreaper/SolisAgileManager/blob/0986ebf9e3dcbe5e24b1100ded2c7fba5cd36471/.github/workflows/ci.yml#L92-L95

Webreaper, 2025-03-04:
Nice spot - could very well be. Testing.... 

@tabannis - could you try pulling the `dev` tag once this build has completed, and let me know if this works?

https://github.com/Webreaper/SolisAgileManager/actions/runs/13647818389

Webreaper, 2025-03-04:
Can you remove the platform line and see if it works?

tabannis, 2025-03-04:
Same result 👎

baruba-cons, 2025-03-04:
Found ["exec format error" running arm64 docker image using dotnet](https://stackoverflow.com/questions/78623677) on SO which might be useful.

@martincostello You've [blogged](https://blog.martincostello.com/look-ma-no-hands-publishing-containers-with-the-dotnet-sdk/) about a new way to publish an app to a container image, do you know if [multi-platform manifest](https://github.com/dotnet/sdk-container-builds/issues/87) is possible using this method?

martincostello, 2025-03-04:
Hey @baruba-cons - sorry, I've not tried to do that myself, so I don't know.

Maybe @baronfel can help you out.

Webreaper, 2025-03-04:
I'm already using `docker/build-push-action` to create the multi-platform manifest. I think the issue might be that the arm64/Pi base image doesn't have `sh` so might need another shell. I wonder if `bash` would work instead?

Webreaper, 2025-03-04:
That new build process looks good, but I'm not sure it's that much different to what I'm already doing. Are there any github actions that leverage it yet?

Webreaper, 2025-03-04:
Actually I see Martin mentions how to use this in a github action, so I could give this a try.

However, I'm *already* publishing multi-arch manifest images to Docker and they all work just fine - except the Arm64 one (which doesn't work on RaspPi). I'd rather understand what's wrong - without having to buy a raspberry pi. 😁

baruba-cons, 2025-03-04:
Since publish uses `--self-contained true`, do you need [.NET Runtime Dependencies](https://hub.docker.com/r/microsoft/dotnet-runtime-deps/)?

See [No such file or directory while running a docker container](https://stackoverflow.com/questions/78969598/no-such-file-or-directory-while-running-a-docker-container) for more details.

baruba-cons, 2025-03-06:
I've pushed a image to docker hub using dotnet publish and it was pretty straightforward.  If you want to try it from your dev machine try the following (ref: [Containerize a .NET app reference](https://learn.microsoft.com/en-us/dotnet/core/containers/publish-configuration)).  Ensure you login with `docker login registry.hub.docker.com` from CLI.

```shell
dotnet publish SolisManager/SolisManager.csproj -t:PublishContainer 
  -p:RuntimeIdentifiers=\"osx-x64;osx-arm64;linux-x64;linux-arm64;linux-arm\" 
  -p:ContainerFamily="alpine" -p:ContainerRegistry="registry.hub.docker.com" 
  -p:ContainerRepository="webreaper/solisagilemanager" -p:ContainerImageTag="dev"
```

Webreaper, 2025-03-06:
Thanks, I'll give that a go. Does the image work on RaspPi?

baruba-cons, 2025-03-06:
I tried it on my project and it worked with an older RPI 3.  Maybe tabannis would be willing to test if you push the dev image tag.

Webreaper, 2025-03-06:
Cool, will try it.

tabannis, 2025-03-06:
I'm certainly willing to give it a go 👍

Webreaper, 2025-03-07:
Okay, tried it, but I'm getting a failure here, and no idea why. 😕  Any ideas @baruba-cons ? At first I thought it might be because it's not building on a Mac host, but even removing the osx targets gives a failure restoring the linux packages.

```
Run dotnet publish SolisManager/SolisManager.csproj -p:RuntimeIdentifiers=osx-x64;osx-arm64;linux-x64;linux-arm64;linux-arm \ -p:Version=1.0.493-dev --self-contained true -p:PublishTrimmed=false \ -t:PublishContainer -p:ContainerFamily=alpine -p:ContainerRegistry=registry.hub.docker.com \ -p:ContainerRepository=***/solisagilemanager -p:ContainerImageTag=alpha
  
  Determining projects to restore...
/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Client/SolisManager.Client.csproj : error NU1102: Unable to find package Microsoft.NETCore.App.Runtime.Mono.osx-x[6](https://github.com/Webreaper/SolisAgileManager/actions/runs/13716190706/job/38361436801#step:7:6)4 with version (= 9.0.2) [/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/SolisManager.csproj]
/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Client/SolisManager.Client.csproj : error NU1102:   - Found 129 version(s) in nuget.org [ Nearest version: 9.0.0-preview.7.24405.7 ] [/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/SolisManager.csproj]
/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Client/SolisManager.Client.csproj : error NU1102:   - Found 0 version(s) in /usr/share/dotnet/library-packs [/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/SolisManager.csproj]
  Restored /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/SolisManager.csproj (in 2.98 sec).
  Failed to restore /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Client/SolisManager.Client.csproj (in 2.98 sec).
  Restored /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Inverters.SolarEdge/SolisManager.Inverters.SolarEdge.csproj (in 2.97 sec).
  Restored /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Shared/SolisManager.Shared.csproj (in 2.97 sec).
  Restored /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Inverters.Solis/SolisManager.Inverters.Solis.csproj (in 1[6](https://github.com/Webreaper/SolisAgileManager/actions/runs/13716190706/job/38361436801#step:7:7) ms).
Error: Process completed with exit code 1.
```

https://github.com/Webreaper/SolisAgileManager/actions/runs/13716245277/job/38361598602

Webreaper, 2025-03-07:
Okay, last problem is auth for the push. Currently I get: 

```
JSDCE minifyWhitespace --exportES6 -o /tmp/emscripten_temp_plh31536/dotnet.native.jso1.js
  SolisManager -> /home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/bin/Release/net9.0/linux-x64/publish/
  Building image '***/solisagilemanager' with tags 'alpha' on top of base image 'mcr.microsoft.com/dotnet/runtime-deps:9.0'.
  Uploading layer 'sha256:7cf63256a31a4cc44f6defe8e1af95363aee5fa75f30a248d95cae684f87c53c' to 'registry.hub.docker.com'.
  Uploading layer 'sha256:3c23d8dbc423f8078d17e36b95bf61e71c982800d38a361e4afb6c658bc590cb' to 'registry.hub.docker.com'.
  Uploading layer 'sha256:9a3805478f07d9a3736b8429fe850131ee3a6324bf91e3fc0a8490472c7519a4' to 'registry.hub.docker.com'.
  Uploading layer 'sha256:27aaca73a128a[34](https://github.com/Webreaper/SolisAgileManager/actions/runs/13717890116/job/38366551311#step:7:35)129101a656e4aa2bf2baedec0739e822eeef5c0063d541b9f' to 'registry.hub.docker.com'.
Error: /usr/share/dotnet/sdk/9.0.200/Containers/build/Microsoft.NET.Build.Containers.targets(252,5): error CONTAINER1013: Failed to push to the output registry: CONTAINER1008: Failed retrieving credentials for "registry.hub.docker.com": No matching auth specified for registry 'registry.hub.docker.com' in Docker config. [/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager/SolisManager.csproj]
Error: Process completed with exit code 1.
```

Don't know what additional stuff I need to do in a github action to authenticate with DockerHub (given I already execute the `login-action` github workflow beforehand). Will have to keep digging.

baronfel, 2025-03-07:
@Webreaper can you check if logging in results in any change to the Docker config file? The .NET SDK reads the Docker config file and parses authentication information from it for each registry you try to push to, and it could be that the content in that file isn't valid for whatever reason? In the abstract this _should_ work, and I have a [sample actions workflow](https://github.com/baronfel/sdk-container-demo/actions/runs/12244073237/job/34155014702) that shows it working, but there of course may be environmental differences or configuration differences in play.

baruba-cons, 2025-03-07:
From [authenticating to container registries](https://aka.ms/dotnet/containers/auth) maybe try setting environment variables  `DOTNET_CONTAINER_REGISTRY_UNAME` & `DOTNET_CONTAINER_REGISTRY_PWORD`?

baronfel, 2025-03-07:
`docker config inspect` doesn't really deal with the docker config/auth file - the file lives at `$HOME/.docker/config.json` by default, but it's possible to configure the system to look in different locations. IIRC the spec is `DOCKER_CONFIG/config.json` for maximal flexibility.

You could also try switching actions to the azure login one (which doesn't use the docker CLI at all), and specifying the registry. I'd love to make this work natively though.

baronfel, 2025-03-07:
> From [authenticating to container registries](https://aka.ms/dotnet/containers/auth) maybe try setting environment variables `DOTNET_CONTAINER_REGISTRY_UNAME` & `DOTNET_CONTAINER_REGISTRY_PWORD`?

please don't rely on those environment variables - they are visible in MSBuild logs and are only supported because VS container publish needs a way to provide credentials to MSBuild.

Webreaper, 2025-03-07:
@baronfel Had a bit of a hectic day at work, so not had any time to look at this. 

Could you point me in the direction of how I'd update the action/workflow to log out changes to the config file? Happy to add what you recommend and then see what the output looks like.

baruba-cons, 2025-03-07:
Not sure if this would make a difference, but reading the documentation `docker.io` can be used instead after .NET SDK 8.0.100.

https://github.com/Webreaper/SolisAgileManager/blob/44a78cfdc283b667d778563209896a654281e5e6/SolisManager/SolisManager.csproj#L14

Webreaper, 2025-03-08:
OMG, setting the registry to docker.io worked!

https://github.com/Webreaper/SolisAgileManager/actions/runs/13735542801

Could somebody with a Pi try pulling webreaper/solisagilemanager:alpha and tell me if it works?

tabannis, 2025-03-08:
Still not running, but the logs now have this error:

`exec /app/SolisManager: exec format error`

tabannis, 2025-03-08:
When I specify docker compose with the platform... it ssys this

`image with reference webreaper/solisagilemanager:alpha was found but its platform (linux/amd64) does not match the specified platform (linux/arm64)`

Is alpha build available in arm64 flavour?

Webreaper, 2025-03-08:
It should be - I specified linux-arm64 on in the `ContainerRuntimeIdentifiers` section. 

@baronfel - any ideas? Seems we have the same issue as I had before using the docker-build-push action....

tabannis, 2025-03-08:
Maybe try adding `linux/arm/v8` in `ContainerRuntimeIdentifiers`?

Webreaper, 2025-03-08:
No such runtime Identifier: https://learn.microsoft.com/en-us/dotnet/core/rid-catalog#known-rids

Perhaps 'linux-musl-arm64' - trying...

Webreaper, 2025-03-08:
That's built. Give it a try.

If that doesn't work it'll have to wait until Monday!

tabannis, 2025-03-08:
Nope :(  oh well. Have a nice weekend (y)

Webreaper, 2025-03-08:
Dammit. Ah well, the garden is calling.

baruba-cons, 2025-03-08:
I'm pretty sure if you change `ContainerRuntimeIdentifiers` to `RuntimeIdentifiers` it'll work, I'm not clear what the former does.

https://github.com/Webreaper/SolisAgileManager/blob/b53fefcd5c9fa1f86f8b98f024454f53ca8b665d/SolisManager/SolisManager.csproj#L13

baronfel, 2025-03-08:
It's a bit complex. `RuntimeIdentifiers` is kind of load-bearing in that `Restore` understands `RuntimeIdentifiers` and will use it to determine if any cross-platform implementation packages should be downloaded at `Restore`-time. So my guidance there is 

> If you a) are making a platform-specific application, and b) know what platforms you would like to target, then always specify all of your supported platforms in `RuntimeIdentifiers` in your applications' project

Secondly, the multi-arch container support works like a logical `foreach` over your `RuntimeIdentifiers`:

```
foreach rid in rids do:
  publish rid-specific container
publish multi-arch container wrapper with each rid-specific container inside of it
```

What this means is that if you specify a RID in `RuntimeIdentifiers` that _isn't valid for the container base image you've chosen_ (for example, any non-Linux RID) then container publish will fail. It's for this reason that we provide `ContainerRuntimeIdentifiers` - it lets you define a subset of `RuntimeIdentifiers` to publish for _when publishing a container only_.

Hope that clears things up a bit.

The workflow that @Webreaper [linked above](https://github.com/Webreaper/SolisAgileManager/actions/runs/13735542801/job/38418751572#step:6:35) is showing a single-RID image only, which _might_ be caused by the use of `--self-contained true`. If I recall, that flag requires that the app be built for a single RID, and so in the absence of a RID on the command line or in the csproj it'll use the RID of the executing platform. My concrete suggestions for @Webreaper:

* remove `--self-contained true` and `-p:PublishTrimmed=false` from your `dotnet publish` command and put this in your csproj instead:

```xml
<PropertyGroup>
  <PublishTrimmed>false</PublishTrimmed>
  <PublishSelfContained>true</PublishSelfContained>
</PropertyGroup>
```

* rename your current `ContainerRuntimeIdentifiers` property to just `RuntimeIdentifiers` and create a new `ContainerRuntimeIdentifiers` property with just `linux-arm64;linux-arm;linux-x64`.
  * This should result in a properly multi-arch image

One thing I also notice - you have both a 'normal' Linux and 'musl' Linux RIDs - we currently do not support pushing a single multi-arch image with mixed libc implementations like this. Typically, most apps would have this as two separate images: one for 'glibc'-based Linux (linux-arm64, linux-arm, linux-x64) and one for 'musl'-based Linux (linux-musl-arm64, linux-musl-x64). You can do this with our tooling if you do two separate publish calls:
  * one with `ContainerRuntimeIdentifiers` set to `linux-arm64;linux-arm;linux-x64`
  * one with `ContainerRuntimeIdentifiers` set to `linux-musl-arm64;linux-musl-x64`

When you do this, you should provide a different tag for the musl-based image. Most people will call this `-alpine` or similar. People split apps into these two tags because the base operating systems can have drastically different behaviors and usually picking musl-based Linux is an explicit choice.

Webreaper, 2025-03-08:
Thanks. I only added the musl ones because I was grasping at straws. I'll remove them.

Can I leave the OSX targets so they'll work for Docker on Darwin?

baronfel, 2025-03-08:
No - on Mac, Docker runs Linux containers. This is also generally the case on Windows, but technically Windows containers are A Thing (just not a thing that is widely used outside of specific back-compat use cases).

Webreaper, 2025-03-08:
Oh, yes, good point. Thanks.

And yeah, Windows is weird, and Windows Docker is even more weird. 🤣

Webreaper, 2025-03-09:
Okay, trying those here: https://github.com/Webreaper/SolisAgileManager/actions/runs/13746674281 ...

Webreaper, 2025-03-09:
The problem with 
```
<PropertyGroup>
  <PublishTrimmed>false</PublishTrimmed>
  <PublishSelfContained>true</PublishSelfContained>
</PropertyGroup>
```
is that I also build non-dockerised builds, which **must** be self-contained. Will the command-line override the csproj in this case?

Webreaper, 2025-03-09:
Okay, that seems fine. 

But the multi-arch still doesn't seem to be working even with those changes - the alpha tag still only has `linux-amd64`.

tabannis, 2025-03-09:
Can confirm this, no change when pulling on my pi5.

baruba-cons, 2025-03-09:
Have you tried it without `--self-contained true`?  Maybe remove those from csproj and yml and try again.  At this point it's a process of elimination and then it can be fine tuned.

https://github.com/Webreaper/SolisAgileManager/blob/a914ea0b67c4cbccb39f2042569bf355f06c87ec/SolisManager/SolisManager.csproj#L9

https://github.com/Webreaper/SolisAgileManager/blob/a914ea0b67c4cbccb39f2042569bf355f06c87ec/.github/workflows/ci.yml#L163

baronfel, 2025-03-09:
If you create a binlog of your publish (add `-bl`) then that can be investigated to find out example what is going on. As I said above, I believe `--self-contained true` on your command line is forcing you into single-RID mode (this is true for all `publish` actions).

Webreaper, 2025-03-09:
I don't have 'self-contained true' on my command line any more, so it can't be that.

Webreaper, 2025-03-09:
Okay, I removed it from the wrong job. Will fix later.

Webreaper, 2025-03-09:
Removed from the dotnet container build job. Still not generating multi-arch for the `alpha` tag.

baronfel, 2025-03-09:
This is _fascinating_. The `PublishSelfContained` flag is forcing the SDK to provide a RID, and that automatically-provided RID is what's blocking the SDK's natural multi-RID container publishing.

If you set `<PublishSelfContained Condition="'$(RuntimeIdentifier)' != ''">true</PublishSelfContained>`, this allows the SDK-driven multi-RID container publish to continue, but at least on my machine I then hit a race condition in the WebAssembly build targets the RID-specific builds then trigger:

```
> dotnet publish -t:PublishContainer -bl
Restore complete (0.4s)
  SolisManager.Shared succeeded (0.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Shared\bin\Release\net9.0\SolisManager.Shared.dll
  SolisManager.Inverters.SolarEdge succeeded (0.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.SolarEdge\bin\Release\net9.0\SolisManager.Inverters.SolarEdge.dll
  SolisManager.Inverters.Solis succeeded (0.3s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.Solis\bin\Release\net9.0\SolisManager.Inverters.Solis.dll
  SolisManager.Client succeeded (4.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Client\bin\Release\net9.0\wwwroot
  SolisManager.Shared succeeded (0.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Shared\bin\Release\net9.0\SolisManager.Shared.dll
  SolisManager.Shared succeeded (0.1s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Shared\bin\Release\net9.0\SolisManager.Shared.dll
  SolisManager.Inverters.SolarEdge succeeded (0.1s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.SolarEdge\bin\Release\net9.0\SolisManager.Inverters.SolarEdge.dll
  SolisManager.Inverters.Solis succeeded (0.1s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.Solis\bin\Release\net9.0\SolisManager.Inverters.Solis.dll
  SolisManager.Inverters.Solis succeeded (0.1s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.Solis\bin\Release\net9.0\SolisManager.Inverters.Solis.dll
  SolisManager.Inverters.SolarEdge succeeded (0.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.SolarEdge\bin\Release\net9.0\SolisManager.Inverters.SolarEdge.dll
  SolisManager.Shared succeeded (0.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Shared\bin\Release\net9.0\SolisManager.Shared.dll
  SolisManager.Client failed with 1 error(s) (2.2s)
    E:\code\nuget-packages\microsoft.net.sdk.webassembly.pack\9.0.2\build\Microsoft.NET.Sdk.WebAssembly.Browser.targets(367,5): error MSB4018:
      The "GenerateWasmBootJson" task failed unexpectedly.
      System.IO.IOException: The process cannot access the file 'E:\Code\Scratch\SolisAgileManager\SolisManager.Client\obj\Release\net9.0\blazor.boot.json' because it is being used by ano
      ther process.
         at Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options)
         at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, Nullable`1 un
      ixCreateMode)
         at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, Nullable`1 unixCr
      eateMode)
         at System.IO.Strategies.FileStreamHelpers.ChooseStrategyCore(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, Nullabl
      e`1 unixCreateMode)
         at System.IO.File.Create(String path)
         at Microsoft.NET.Sdk.WebAssembly.GenerateWasmBootJson.Execute()
         at Microsoft.Build.BackEnd.TaskExecutionHost.Execute()
         at Microsoft.Build.BackEnd.TaskBuilder.ExecuteInstantiatedTask(TaskExecutionHost taskExecutionHost, TaskLoggingContext taskLoggingContext, TaskHost taskHost, ItemBucket bucket, T
      askExecutionMode howToExecuteTask)
  SolisManager.Client succeeded (4.5s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Client\bin\Release\net9.0\wwwroot
  SolisManager.Client succeeded (4.5s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Client\bin\Release\net9.0\wwwroot
  SolisManager.Inverters.Solis succeeded (0.1s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.Solis\bin\Release\net9.0\SolisManager.Inverters.Solis.dll
  SolisManager.Inverters.SolarEdge succeeded (0.2s) → E:\Code\Scratch\SolisAgileManager\SolisManager.Inverters.SolarEdge\bin\Release\net9.0\SolisManager.Inverters.SolarEdge.dll
  SolisManager succeeded (12.5s) → bin\Release\net9.0\linux-arm64\publish\
  SolisManager succeeded with 1 warning(s) (32.6s) → bin\Release\net9.0\linux-arm\publish\
    E:\Code\Scratch\SolisAgileManager\SolisManager\Services\InverterManager.cs(28,12): warning CS8618: Non-nullable field 'inverterAPI' must contain a non-null value when exiting constructor. Consider adding the 'required' modifier or declaring the field as nullable.
  SolisManager failed (3.0s) → bin\Release\net9.0\publish\

Build failed with 1 error(s) and 1 warning(s) in 55.1s
```

Webreaper, 2025-03-10:
Okay, merged the PR, but the build failed: https://github.com/Webreaper/SolisAgileManager/actions/runs/13758662323/job/38470180724

```
  Optimizing assemblies for size may change the behavior of the app. Be sure to test after publishing. See: https://aka.ms/dotnet-illink
/usr/share/dotnet/sdk/9.0.200/Sdks/Microsoft.NET.Sdk.StaticWebAssets/targets/Microsoft.NET.Sdk.StaticWebAssets.Compression.targets(366,5): error : The process cannot access the file '/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Client/obj/Release/net9.0/blazor.publish.boot.json' because it is being used by another process. [/home/runner/work/SolisAgileManager/SolisAgileManager/SolisManager.Client/SolisManager.Client.csproj]
```

Also seems to have tagged the image wrong: `Uploading tag 'alpha-linux-arm64' to 'docker.io'.`

Doesn't feel like we're getting any closer. 😄 Any ideas?

Webreaper, 2025-03-10:
Okay, maybe a red herring. Just run a couple of pipelines for unrelated fixes, and it's built. So some race condition (I think you mentioned earlier @baronfel).

The `alpha` tag is now showing as multi-arch - so @tabannis do you want to give it a go?

[image]

tabannis, 2025-03-10:
That's sounding promising. However... I've just tried an image pull and container recreation remotely, but something had gone wrong and I've lost my vpn (Wireguard) home. Doh. 

I'll have to check when I get home after work to see what's happened. Bear with me caller.

Webreaper, 2025-03-10:
LOL. Anyone else hate technology? 😄

baronfel, 2025-03-10:
Confirmed that the multi-arch manifest looks good, however the generated containers have the wrong Entrypoint:

```
"Entrypoint": [
                "dotnet",
                "/app/SolisManager.dll"
            ]
```

the container base image used (runtime-deps) doesn't include `dotnet` itself, because it's used as the base for self-contained applications. I _suspect_ that the shenanigans we had to do to prevent the win-x64 RID from being used in the outer build has impacted the way we compute which entrypoint to use. Will take a look.

baronfel, 2025-03-10:
Ok, with https://github.com/Webreaper/SolisAgileManager/pull/137 I've fixed this for good - I've got the app running locally in a container, no problem.

[33 more replies]