Webreaper/SolisAgileManager #128: IOG History - show reduced rate when intelligent chargin

dqj999, 2025-03-04:
Currently, for daytime periods using Intelligent charging with IOG, the cost shown in the History table view shows the full price rather than the 6p rate. This despite the comment recognising that the car is being charged.

[image]

Webreaper, 2025-03-04:
Yeah, it's been flagged before. I don't believe there's an API available where I can get the IOG smart-charge price. But anyway, that price is showing the price of the 30-minute tariff slot, not the price of the smart-charge.

Might look at this in future, but it depends what Octopus provides on their API.

dqj999, 2025-03-04:
> I don't believe there's an API available where I can get the IOG smart-charge price.

I realise that it might change but Octopus say that any Smart Charge is at the overnight rate which you already know and use.

Webreaper, 2025-03-04:
Will have a look.

Webreaper, 2025-03-04:
Committed a change, it might do the trick. Can you pull the latest dev build and give it a try?

Webreaper, 2025-03-04:
This went into the release, I'll close this issue. Please reopen if it didn't work. :)

dqj999, 2025-03-04:
Thanks. I'll have to wait for some daytime EV charging in order to check this out.

dqj999, 2025-03-11:
[image]

I’ve just checked what this fix does and unfortunately it applies the 7.00p value to the wrong column, specifically the solar power estimate!

Webreaper, 2025-03-11:
Brilliant. D'oh!

Webreaper, 2025-03-11:
Pushed a fix to dev.