Webreaper/SolisAgileManager #129: Battery Charge continued after "Boost" slot ended.

tabannis, 2025-03-04:
Last night (in the middle of the night) my SOC dropped below 35%, as this the threshold for "% for peak period"  and peak was due a boost slot was called for, however, rather than just boosting for the 1x30 minute slot, the SOC was taken to 100% over multiple slots including 4 marked as "Do nothing". 

[image]

Is this by design? Or am I misunderstanding what "boost" does, in that i assumed it would only "top up" the battery rather than fully charge it.

Logs and history below, 

[SolisManager-20250304.log](https://github.com/user-attachments/files/19070117/SolisManager-20250304.log)
[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/19070116/SolisManagerExecutionHistory.csv)

Webreaper, 2025-03-04:
Interesting. It should have only boosted the orange column...

tabannis, 2025-03-04:
Yeah, looking at the logs myself, there's no reference at all to the boost charge. However, Look what is in my charge slots according to solis cloud...

[image]

tabannis, 2025-03-04:
...That appears to be the charge slot from 03/03.

[SolisManager-20250303.log](https://github.com/user-attachments/files/19070458/SolisManager-20250303.log)

`[01:30:01.480--INF] Sending new charge instruction to Solis Inverter: 50, 0, 01:30-04:00, 00:00-00:00`

Webreaper, 2025-03-04:
Okay, looking at the logs, I can't see any indication that SolisManager set your inverter to charge at all in the whole of that log. All of the log entries are:
```
[04:00:02.102--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
```
which implies that a) we tried to set the inverter to 'do nothing', and b) the inverter was already in that state, so no changes were made to the inverter.

Are you sure that you don't have anything else attempting to control the inverter at the same time? And are any of the other charge/discharge slots set to anything? I can't see anything in the log that would have caused that.

Webreaper, 2025-03-04:
Oh, our messages crossed. That would make sense. I can only assume that this is caused by the problems everyone has been seeing with the solis API in the last week or so. i.e., I'm asking the inverter what it's state is, and it's not returning the _actual_ state of the inverter, but some stale data.

What does the status line at the bottom of the charging table say?

tabannis, 2025-03-04:
So I just switched the current slot we are in to "charge" and the inverter was updated. Switched it back

```
[12:25:47.418--INF] Updating slot action for "2025-03-04T12:00:00.0000000Z" to "Charge"...
[12:25:47.422--INF] Set override: 12:00 - Charge
[12:25:47.424--INF] Sending new charge instruction to Solis Inverter: 75, 0, 12:00-12:30, 00:00-00:00
[12:26:37.946--INF] Updating slot action for "2025-03-04T12:00:00.0000000Z" to "DoNothing"...
[12:26:37.946--INF] Cleared override: 12:00 - DoNothing
[12:26:37.947--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
```

The 12:00-12:30 slot went up to the inverter OK, but it hasn't cleared it out...

[image]

Adds weight to your theory, but unsure if this is just a "me" problem or not.

Webreaper, 2025-03-04:
I've found this. A big in the simulation/testing logic meant that sometimes the inverter state wasn't being read and it was using the last value. Should be fixed in the build that's running now. Will test and confirm, and then do a release.

tabannis, 2025-03-04:
Updated and everything looks good. Thanks for jumping on this one so quick.

Webreaper, 2025-03-04:
It was just lucky you raised it. I'd noticed some oddness in my logs, but was planning to look later... 👍