Webreaper/SolisAgileManager #132: Not using cheapest rates

By-tor, 2025-03-04:
The changes over the past couple of weeks (while I was away on holiday) seem to have made some great improvements to the calculations.
I still, however, have reservations over when it chooses to charge though.
The strategy for tomorrow on my system shows this:

[image]

As you can see, the system is looking to boost during the night when the price is around 16p - that will be OK as long as it doesn't fill the battery to 100%.

Tomorrow morning, though, it will charge 5 slots at 9p to 15p which will definitely fill the battery leaving it too full to take advantage of the next 4 slots 5p to 6p.

Does the the app calculate the running cost of what is stored in the battery? 
e.g.  2 slots at 20p giving 3Kwh at a cost of 60p
        4 slots at 5p  giving 6Kwh  at a cost of 30p
        Average cost of power in battery 10p per Kwh
If not, cold this be used as part of the decision of whether to charge or not? i.e. is it more advantageous to use the battery or grid?

I also do not get paid for any export (rent a roof scheme) so I do not get any benefit from FIT. Do I need a different strategy?

Webreaper, 2025-03-04:
What do you have the "slots needed to fill the battery" set to? Is it set at 6?

By-tor, 2025-03-04:
[image]

Webreaper, 2025-03-04:
Okay, I've had a look. It's actually pretty simple to see what's going on. If you move your mouse over the plan you'll see a tooltip pop up which shows you the reason why it's doing what it's doing in any given slot. In this case, it's because you have the "Always charge below (p/kWh)" setting set to 15. This means that regardless of SOC, charge state, peak or anything else, if a slot is below 15p, it will **always** charge the battery. 

[image]

So the 11 slots that are charging are all under 15p. 

[image]

If you don't want it to do that, just lower the value to something like 5p. You'll see that the charging strategy immediately converts all those slots to Boost slots, which will only charge if the battery drops below 50% (based on your setting):

[image]

To answer your other question:

> I also do not get paid for any export (rent a roof scheme) so I do not get any benefit from FIT. Do I need a different strategy?

It depends what you want to do. There's no different strategy for exporting, only different settings. So for example, for me, if I export, I get 15p/kWh. So if the price drops below 15p (usually around 13-14p, after some inverter/battery losses for charging and discharging) I _always_ want to charge - because even if it's sunny, I'd rather charge from the grid at 13p, fill the battery, and then export any excess at 15p.

For you, if you don't have any export, you should probably only set it to 'always charge' if the price goes negative, or very low. But that's up to you - play with the settings and see what works for you. Also, I'd recommend getting onto a tariff that allows you to export if you can otherwise you're throwing money away for any excess energy you generate from PV that you can't use via your house load or battery.

closed 2025-03-06 by Webreaper