Webreaper/SolisAgileManager #138: Running in Docker on macOS

0rangutan, 2025-03-10:
Has anyone got this running successfully in Docker on macOS?
I have followed the guidance provided in the release and the docker-compose file, but am seeing these errors when I try to run the container:

```
solismanager exited with code 255
solismanager  | exec /app/SolisManager: no such file or directory
```
Any ideas?  Thanks

Webreaper, 2025-03-10:
Have a look at the last 3 comments in #124. The alpha tag, with the environment variable in the docker-compose snippet, should work.

Webreaper, 2025-03-11:
Closing this - the latest image should work (you won't need the env var). Please let me know if it does/doesn't.

closed 2025-03-11 by Webreaper

0rangutan, 2025-03-11:
Thanks - I have the app running on macOS now!
Works with the Alpha and currently with 1.0.522 and the Env variable.
I'll try it without now...

0rangutan, 2025-03-11:
Yes, works without the environment variable, thanks!