Webreaper/SolisAgileManager #139: Further issue with detecting Ohme charging

dqj999, 2025-03-11:
The. Updated Ohme software does not let you avoid dynamic charging now so your car may charge randomly overnight if it is plugged in. I don’t think that any notification is sent when this happens so you could see your house battery being used to charge the car overnight. I can’t think of a way round this with software so I think it will have to raised as a bug/problem with Ohme.

dqj999, 2025-03-11:
Having thought about this you could use the scheduled action facility to reduce the effect of this. If you set a low power scheduled charge for say the first 4 hours of the cheap period then a high power charge in the last two hours that would correct any clash between the two charging systems and would ensure that the battery was at the desired charge level in the morning, although it might have had a few random charge/discharge cycles.

Webreaper, 2025-03-11:
Yeah, I was going to suggest that if you set a scheduled action to charge the battery all the way through the cheap overnight period (which you'd probably want anyway) then you could prevent the battery discharging to charge the car.

dqj999, 2025-03-11:
Yes thanks,  I spotted that just after I made the first comment. Good feature!

Webreaper, 2025-03-12:
No, it'll work for any version of the firmware.

closed 2025-03-12 by Webreaper