Webreaper/SolisAgileManager #140: Schedule Action set once to Inverter and retained

cs95dtt, 2025-03-11:
Was wondering about the Scheduled Actions specifically for charging actions say 23.30-05.30 in my example where I want to guarantee an overnight charge no matter what.   Is it possible to have this set once to the inverter without it being reset?  So, in the case Solis Agile Manager webserver has an outage (host failure for example) I can be sure an overnight charge will always happen.

I had had a look at setting this directly at the Inverter using the secondary or third charging periods (leaving the first period free for Solis Agile Manager to utilise) but this causes a time overlap conflict if the Solis Agile Manager tries to apply charging periods.

Thanks for the amazing work on this project and will buy you coffees for such brilliant and simple to use solution.

cs95dtt, 2025-03-11:
I just had a thought the amp value is reset to 0 so any sort of permanent override wouldn't work anyway for 23:30-05:30 charge period I want to set permanently.

I'm over thinking and complicating this.

Great work nevertheless from you!

Webreaper, 2025-03-12:
Yeah, mixing manual SolisCloud updates with the app becomes complicated because of the potential for conflicts, which then stop the app working correctly. The app pretty much blats over the entire charging setup when it applies the charge state for a new slot, to avoid this. 

I think if you just set up a bunch of 'Charge' scheduled actions it should do what you want, though, right? The fact that the 23:30-05:30 charge will be written to the inverter once, at 23:30, each day isn't really a biggie.

closed 2025-03-12 by Webreaper