Webreaper/SolisAgileManager #143: Low battery didn't charge fully overnight

tabannis, 2025-03-14:
For some reason last night my battery didn't charge fully despite having the slots available.

[image]

looking at the logs, the inverter was sent the charge instructions as it dropped below 25%, but 5 minutes later canceled it? This repeated during the charge "window".

Filtered log snippets...

```
[02:01:00.479--INF] Refreshed state: SOC = 21%, Current PV = 0.000kW, House Load = 0.244kW, Forecast today: 12.44635kWh, tomorrow: 11.68290kWh
[02:05:00.873--INF] Sending new charge instruction to Solis Inverter: 95, 0, 02:00-02:30, 00:00-00:00
[02:09:00.476--INF] Refreshed state: SOC = 23%, Current PV = 0.000kW, House Load = 0.297kW, Forecast today: 12.44635kWh, tomorrow: 11.68290kWh
[02:10:01.386--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
[02:14:00.474--INF] Refreshed state: SOC = 23%, Current PV = 0.000kW, House Load = 0.233kW, Forecast today: 12.44635kWh, tomorrow: 11.68290kWh
```

```
Line  77: [02:05:00.873--INF] Sending new charge instruction to Solis Inverter: 95, 0, 02:00-02:30, 00:00-00:00
	Line  79: [02:10:01.386--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
	Line  88: [02:30:01.203--INF] Sending new charge instruction to Solis Inverter: 95, 0, 02:30-04:00, 00:00-00:00
	Line  92: [02:35:01.603--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
	Line 102: [03:00:01.691--INF] Sending new charge instruction to Solis Inverter: 95, 0, 03:00-04:00, 00:00-00:00
	Line 106: [03:05:01.089--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
	Line 116: [03:30:01.188--INF] Sending new charge instruction to Solis Inverter: 95, 0, 03:30-04:00, 00:00-00:00
	Line 120: [03:35:01.020--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
	Line 135: [04:05:00.825--INF] Sending new charge instruction to Solis Inverter: 95, 0, 04:00-04:30, 00:00-00:00
	Line 137: [04:10:00.899--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
	Line 146: [04:30:01.719--INF] Sending new charge instruction to Solis Inverter: 95, 0, 04:30-05:00, 00:00-00:00
	Line 150: [04:35:01.924--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
	Line 180: [05:45:00.881--INF] Sending new charge instruction to Solis Inverter: 95, 0, 05:30-06:00, 00:00-00:00
	Line 182: [05:50:01.578--INF] Sending new charge instruction to Solis Inverter: 0, 0, 00:00-00:00, 00:00-00:00
```

[SolisManagerConfig.json](https://github.com/user-attachments/files/19242524/SolisManagerConfig.json)
[SolisManager-20250314.log](https://github.com/user-attachments/files/19242512/SolisManager-20250314.log)
[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/19242513/SolisManagerExecutionHistory.csv)

Webreaper, 2025-03-14:
That's a bit of an odd one! Will take a look

Webreaper, 2025-03-14:
(edited to remove the config file, that shouldn't be shared because it has your API keys in it).

tabannis, 2025-03-14:
Oops. Yeah, my bad 🤦‍♂️

tabannis, 2025-03-14:
Ah, ok. (that'll teach me for running the dishwasher on a spenny cycle before bed 😂)

That makes a lot of sense. No rush for me lots of free sun predicted for the next few days ☀️

Happy to test 👍

Webreaper, 2025-03-14:
Technically the 'charge if battery below x' is working as expected. The real question is why those slots were set to be "Boost" rather than "Charge". They should have been marked as 'cheapest' and then used to charge.

tabannis, 2025-03-14:
Yeah, it's a bit odd.

The 2am to 2.30am slot in the history is marked as "DoNothing" but the logs have charge instructions in them🤔

Webreaper, 2025-03-14:
Okay, I've done quiite a big refactor which should fix this. Now if the battery drops below the threshold SOC, it will change the slot action for the full 30 minutes, and won't revert to 'Do Nothing' as soon as the SOC gets above the threshold. 

However, it's changed quite a few things so will need careful testing. This change will also mean that scheduled actions and IOG Smart Charge slots can be manually overriden, which wasn't possible before. 

https://github.com/Webreaper/SolisAgileManager/actions/runs/13856166046

@dqj999 and @martinpike69 - would you be able to test this dev build and check to see I haven't broken IOG smart-charging slots?

martinpike69, 2025-03-14:
I'm pretty convinced that Octopus aren't 100% reliable in sending the Smart 'slots'. Not a lot you can do about that though!

@dqj999 how is yours going?

Webreaper, 2025-03-14:
Yeah, they're not. But this isn't isn't about IOG - it's a different issue.

 I've made a lot of changes - they should work, but I have no way to test the IOG parts. So just want to validate that the existing functionality works as it did before, even if Octopus aren't consistent in sending the slots.

tabannis, 2025-03-14:
Thanks! I've spun up a Dev container and put "latest" in simulation mode. I'll see how it gets on with Agile.

tabannis, 2025-03-14:
I'll see what the prices look like later and get the battery in a state that needs a good charge overnight and see what occurs.

dqj999, 2025-03-14:
I’m finding it very useful and I usually upgrade to the latest dev build. I did a Smart Charge last night and it set the correct slots to charge the house battery. 
The Ohme app seems fairly useless at staying connected to my Nissan Leaf which means that a charge to 90% can get reinterpreted as a request to add 90% resulting in a lot of heavy charging early on.
I’ve decided that the best strategy for me on IOG is to have a scheduled charge to about 60% overnight and then let my solar panels take the battery up to 100% by noon. Of course I would make more money by charging the battery to 100% every night and then exporting the power I bought at 7p at the Export tariffs rate of 15p. This thrashes the battery however.

Webreaper, 2025-03-14:
> Just updated to this dev and found the app immediately set a charge slot even though my SOC was over the set threshold
> Is this correct due to some other override?

Nice spot. I think that's because it's evaluating the plan before it's received any SOC from the battery, so it thinks "Ooh, battery is low, better charge it". Can you click on the plan action (i.e., charge) and send me a screenshot of the explaination, like this?

[image]

I presume it'll say "SOC is below threshold" but I just want to check. In the meantime I'll put in a fix.

By-tor, 2025-03-14:
sorry about the size - can't capture a pop-up on PC so had to screenshot phone :-)

[image]

Webreaper, 2025-03-14:
No prob. 

So that's correct then - you have the settings set to aim for 60% before the peak, and 15:30 is before the peak. So it's charging at the cheapest rate before the peak to try and get the battery to the required level. It's working as expected. 😄

By-tor, 2025-03-14:
Is it using SOC is zero ? 
This would trigger charge if SOC is below 10% ?

[image]

Webreaper, 2025-03-14:
Yes, that's probably it. It's a race condition at startup - it tries to evaluate the plan rules before the inverter state has been successfully queried. I've change it so that the first plan evaluation at startup will force an inverter status update, so it should never try and recalc with Battery SOC == 0. Should fix it. 

https://github.com/Webreaper/SolisAgileManager/actions/runs/13860248175

tabannis, 2025-03-14:
[image]

Interestingly my solis graph matches @By-tor from last night, we obviously have different charge parameters so I wonder if the slot prices were extra confusing for the algo? 

Meh, I've put .541.0 into service 👌

Webreaper, 2025-03-14:
Yeah, same bug will affect all of you. ;)

I didn't notice it because on Cosy, I get 2 hours of charging every day at 10pm, and I only run my dishwasher while it's cheap and the battery is charging.... 😄 

Changes today should significantly improve things though now, and also fix a couple of other things that have been nagging me. So let's see how it runs overnight, and if it looks okay, I'll do a release tomorrow evening.

Webreaper, 2025-03-15:
Looks good. The reason it didn't charge too much is that you have it set to "35% to get you through the peak". So once it's over 35% in the cheapest period, it won't do any more charging. If you want it to boost more, ramp that number up and it'll add in more Pre-peak slots and get the battery higher before the rise in prices.

So I think it's working correctly now. Let's monitor for another day or so, and see if the IOG stuff still works (ie I haven't broken it).

By-tor, 2025-03-15:
Worked well overnight for me, didn't charge during the night as forecast for today is good. 
Charged using two full slots as SOC was below 50%

[image]

[image]

On the subject of not charging overnight when solar forecast is good, would it be worth changing that from a binary option to a SOC target - e.g. if forecast is over 9Kwh charge to 40% SOC (even better if it could be inversely proportional SOC target related to forecast)

Webreaper, 2025-03-15:
Yeah, I'm still figuring out the best (but also simplest) strategy for dealing with forecast and how it should change what happens. I should open an issue and solicit everyone's ideas and see what works best.

Webreaper, 2025-03-19:
Closing this because the original issue is fixed.

closed 2025-03-19 by Webreaper