Webreaper/SolisAgileManager #148: Ohme Smart Slots not coming through?

martinpike69, 2025-03-16:
Plugged my car in 1932. Got given a low rate slot immediately until 1033 next day.

Now at 1947 and no low rate slots have come through.

Appreciate there isn't a lot you can do about this.

Will now charge via Manual Actions.

Log attached.

[SolisManager-20250316.log](https://github.com/user-attachments/files/19274598/SolisManager-20250316.log)

Webreaper, 2025-03-17:
Yeah, nothing I can do about that. But if you do get a notification for a slot come through please confirm to ensure I haven't broken anything for when it _does_ work. :)

closed 2025-03-17 by Webreaper

Webreaper, 2025-03-18:
I could do, but it would fill up the logs a bit. I can certainly add something temporarily.

Webreaper, 2025-03-18:
Try this build here: https://github.com/Webreaper/SolisAgileManager/actions/runs/13916813891

Once you've run it for a week or so let me know and I'll remove the logging.

One plan I have is to have different levels of logging and a control in the app so you can enable more detailed logging - at which point I could leave this in.

martinpike69, 2025-03-19:
My car just started charging at 1141, unprompted by me - it was just sat there still plugged in from last night.

Smart Slot has arrived from Octopus.

Looks to me as if 'instant' Smart Slots, i.e. those that start immediately you authorise them via the Ohme app, don't get sent via the API, but those where some time has elapsed, may do.

I'll experiment further by asking for a lower % to be added, thus hopefully not receiving an 'instant' slot.

dqj999, 2025-03-19:
Sometimes these initial charges only last for a few minutes and no notification seems to be raised. 

The other thing is off you leave your car plugged in for more than 24 hours another charge cycle can start. If it was set to add say +20% then another 20% charge can be added.