Webreaper/SolisAgileManager #149: Add inverter temp alert

Webreaper, 2025-03-16:
Thermometer and alert if temp is too high.

tabannis, 2025-03-17:
Or alert if too low? My battery and inverter live outside, and the battery frequently gets below it's ideal operating temperature window in the winter and shuts down, so perhaps a temperature range? I know the API doesn't return battery temp, but it would still be good to know if the inverter is a bit chilly as an early warning.

Webreaper, 2025-03-18:
Yep, that makes sense. It's only going to be an alert in the app, not a push alert, as that's much more complex. But better than nothing. I'm reorganising the room my inverter is in, and want to ensure the airflow and temp are adequate.