Webreaper/SolisAgileManager #151: Multiple installations - odd behaviour

By-tor, 2025-03-17:
I have a proxmox server running several virtual machines (for several years) and I have created a windows 10 VM, installed SolisAgileManager configured all the requirements and have been running and updating it for a few weeks with a few issues that have been worked through and the system and settings are beginning to behave as required with lots of support from you guys. Thank you.

I have today cloned that VM, as there is a second site installation I also wish to control. This is a process I have done several times successfully in the past for other applications. The cloned VM has different settings for Octopus, Solcast and Soliscloud API's

I have two issues which are weird.

When running concurrently, the two VM's seem to have some interaction and conflict with each other even though they have separate IP's and Macs.

When the doner VM is run alone, there is no issue.

When the cloned VM is run alone, the charging plan is from today to the end of March i.e. 15 days long.

Any ideas anyone?

Webreaper, 2025-03-18:
Assuming that the instances of the app are running on completely separate VMs, and they are storing their config and data in two different separate folders, I can't possibly see how they could interact with each other.

Do both instances have separate API keys for the inverter?

The length of the future charging plan is defined by the number of future slots that Octopus returns for the selected tariff. Which tariff are you selecting?

dqj999, 2025-03-18:
I wouldn’t have thought that it was a good idea to run two instances- unless they were working on completely different inverters with completely different accounts on Octopus, Solis and Solcast. Is that what you are describing?

Webreaper, 2025-03-18:
So I've played with this a bit, because I have my 'production' instance (my actual solar) and my dev instance. The latter is usually running in simulate-mode, which means it reads from the inverter, but never writes. In general it's fine if you're doing that. As you say, if you have multiple instances pointing at the same inverter and trying to update, it's all going to get a bit conflicted - but I don't think that's what @By-tor is doing (I hope). 

In terms of Octopus, that shouldn't make any difference at all, because the Octopus API calls are all entirely read-only. In fact, apart from the IOG smart-charge check, you don't even _need_ an Octopus account number/key to make the calls - the tariff price APIs are all public. The Octopus API key is really only there so I can query your specific tariff and see if it changes (and for IOG slots if you're using them).

By-tor, 2025-03-18:
Yes these are working on completely different inverters with completely different accounts on Octopus, Solis and Solcast. 

I think the issue may be with the cloning process, I will need to run windows sysprep and make a template and then clone that.

Not sure what is going on with the 15 days of charging plan though - that is a separate issue I think

Webreaper, 2025-03-18:
> Not sure what is going on with the 15 days of charging plan though - that is a separate issue I think

What tariff are we talking about? If it's something like a fixed tariff, it's possible that Octopus provide longer-term tariff data. I did notice when I was building the comparison page that for some tariffs they return days of data - so I truncate to make the comparison sensible.

By-tor, 2025-03-18:
Tarrif is E-1R-VAR-22-11-01-D

Webreaper, 2025-03-18:
Okay, yes - so this is down to Octopus. If you get the tariff data from here: https://api.octopus.energy/v1/products/VAR-22-11-01/electricity-tariffs/E-1R-VAR-22-11-01-D/standard-unit-rates/ you get rates for several months. Most of it is historic data, but the latest rate goes to 30-March. 

Now, because the charging plan has to work in 30-minute increments, I take the Octopus tariff data and split it up into 30 minute slots. So you get a _lot_ of slots. Here's the log:
```
[11:58:28.242--INF] Retrieved 2 rates from Octopus (01-Jan-2025 00:00 - 31-Mar-2025 23:00) for product E-1R-VAR-22-11-01-D
[11:58:28.248--INF] 1294 new Octopus rates available to 31-Mar-2025 23:00 (cheapest: 25.75818p/kWh, peak: 26.498955p/kWh)
```
So I think what I need to do is cap it at, say, 96 slots - so the next 2 days. Otherwise the UI gets extremely sluggish.... 

Pushing a fix to dev which will do this.

Webreaper, 2025-03-19:
Closing this because the 15-day issue is fixed. Don't think I can fix the other weirdness!

closed 2025-03-19 by Webreaper