Webreaper/SolisAgileManager #152: Additional IOG dispatch slot appears which should be there

cs95dtt, 2025-03-18:
I've noticed IOG dispatch slots are correctly retrieved according to the log file but for some reason extra slot is applied which shouldn't be there which I can't workout why.

I have config that will charge the solar battery if the price drops below 8p which so the IOG slots should always charge.

Most recent example was today, IOG slots 11:00-11:30 and 11:30-12:00 but for some reason it applies 3 slots for smart charge, the extra slot is 10:30-11:00.

[10:23:01.098--INF] Found 2 IOG Smart-Charge slots (out of a total of 2 planned and 0 completed dispatches)
[10:23:01.102--INF] SmartCharge Dispatches:
  Time: 11:00 - 11:30, Type: smart-charge, Delta: -0.18
  Time: 11:30 - 12:00, Type: smart-charge, Delta: -3.60
[10:23:01.108--INF] Applying charge action to 3 slots for IOG Smart-Charge

I've check with both Windows version of Solis Manager 1.0.468 and 1.0.544 both exhibit the same issue.

Attached is the Ohme charger screenshot showing the 11:00-12:00 slot.
[image]

Also screenshot of Solis Manager showing the additional 10:30-11:00 slot.
[image]

I've also attached the log and the config (minus my inverter and account details):
[SolisManagerConfig.json](https://github.com/user-attachments/files/19325864/SolisManagerConfig.json)
[SolisManager-20250318.log](https://github.com/user-attachments/files/19325865/SolisManager-20250318.log)

As mentioned the log shows the two slots detected but for some reason 3 slots are applied - I've not been able to ascertain what is causing this and wondered if this is a bug as the IOG query appears to be retrieving the correct slots.

Webreaper, 2025-03-19:
Looks like a `>=` should have been a `>`. I've pushed a fix - can you try the latest [dev build](https://github.com/Webreaper/SolisAgileManager/actions/runs/13940884056)? If it looks good, I'll push a proper release.

cs95dtt, 2025-03-19:
Thanks for the prompt reply.  Testing the DEV instance and will let you know if a couple of days of the outcome.

closed 2025-03-20 by Webreaper