Webreaper/SolisAgileManager #155: Price column on history table view inaccurate

dqj999, 2025-03-22:
The price per unit column on the tabular view of the history is inaccurate. It gives a very false impression of the cost being paid per unit when exported electricity is generated. Since (currently) the app has no idea of the export tariff this is hardly surprising but a quick fix would be to use negative values for the cost of exported electricity. This would overestimate the contribution of exports but would not as currently happens increase the recorded cost of the power. Fixing this properly would require the export tariff information to be taken into account, possible I suppose but needing more work.

Webreaper, 2025-03-22:
That isn't what that column is for. It's not trying to calculate what you have or haven't paid for electricity at the time. It's simply a list of what the octopus price is for the current tariff, during each 30 minute period.

Trying to calculate what you've been billed would need to take account of import/export every minute or so, and would be either very complex or very inaccurate.

This app is only designed to optimise your battery charging schedule. It's not a billing management app, and I'm not interested in building or supporting that - if people want to track their exact costs they can either use the octopus app (with a Home Mini) or use one of the many other apps that focus on this sort of billing accuracy (OctopusWatch, Octopus compare, Home Assistant etc etc).

closed 2025-03-22 by Webreaper

dqj999, 2025-03-22:
OK understood.