Webreaper/SolisAgileManager #156: IOG dispatch slots change mid-slot but previous slots not stopped

cs95dtt, 2025-03-24:
Hi.

I've noticed that when IOG dispatch updates are correctly retrieved but when IOG slots change mid-slot the existing slot isn't cancelled/reset even when the car stops charging and more than 10-15mins has elapsed.

Example below is 21:30-22:00 IOG slot is applied but 5mins later at 21:35 the IOG slots update and that previous slot is no longer at the cheap rate and instead new slots 22:00-22:30 onwards are applied to inverter but the original slot is not cancelled/reset to do nothing.  Do the regular 5mins checks stop any existing slots that no longer valid and the car stops charging?    In the interim I've been manually overriding these slots that don't get cancelled.

[21:30:14.903--INF] Checking for IOG Smart Charge slots....
[21:30:15.056--INF] Found 2 IOG Smart-Charge slots (out of a total of 2 planned and 0 completed dispatches)
**[21:30:15.066--INF] SmartCharge Dispatches:
  Time: 21:30 - 22:00, Type: smart-charge, Delta: -0.54**
  Time: 22:30 - 05:30, Type: smart-charge, Delta: -50.57
[21:30:15.072--INF] Applying charge action to 15 slots for IOG Smart-Charge
[21:30:18.795--INF] Inverter already in correct state (10, 0, 21:30-22:00, 00:00-00:00) so no charge instructions need to be applied
[21:34:00.923--INF] Refreshed state: SOC = 22%, Current PV = 0.001kW, House Load = 0.848kW, Forecast today: 1.43665kWh, tomorrow: 11.42975kWh
[21:35:01.207--INF] Checking for IOG Smart Charge slots....
[21:35:01.877--INF] Found 3 IOG Smart-Charge slots (out of a total of 3 planned and 0 completed dispatches)
**[21:35:01.890--INF] SmartCharge Dispatches:
  Time: 22:00 - 22:30, Type: smart-charge, Delta: -3.61**
  Time: 22:30 - 23:00, Type: smart-charge, Delta: -0.54
  Time: 23:00 - 05:30, Type: smart-charge, Delta: -46.96
[21:35:01.898--INF] Applying charge action to 15 slots for IOG Smart-Charge
[21:35:02.880--INF] Sending new charge instruction to Solis Inverter: 10, 0, 21:30-05:30, 00:00-00:00
[21:39:01.186--INF] Refreshed state: SOC = 23%, Current PV = 0.001kW, House Load = 0.851kW, Forecast today: 1.43665kWh, tomorrow: 11.42975kWh
[21:40:01.026--INF] Checking for IOG Smart Charge slots....
[21:40:01.240--INF] Found 3 IOG Smart-Charge slots (out of a total of 3 planned and 0 completed dispatches)
**[21:40:01.249--INF] SmartCharge Dispatches:
  Time: 22:00 - 22:30, Type: smart-charge, Delta: -3.61**
  Time: 22:30 - 23:00, Type: smart-charge, Delta: -0.54
  Time: 23:00 - 05:30, Type: smart-charge, Delta: -46.96
[21:40:01.255--INF] Applying charge action to 15 slots for IOG Smart-Charge
[21:40:03.146--INF] Inverter already in correct state (10, 0, 21:30-05:30, 00:00-00:00) so no charge instructions need to be applied
[21:43:33.335--INF] Updating slot action for "2025-03-23T21:30:00.0000000Z" to "DoNothing"...
[21:43:33.347--INF] Set override: 21:30 - DoNothing
[21:43:33.358--INF] Checking for IOG Smart Charge slots....
[21:43:33.537--INF] Found 3 IOG Smart-Charge slots (out of a total of 3 planned and 0 completed dispatches)
**[21:43:33.541--INF] SmartCharge Dispatches:
  Time: 22:00 - 22:30, Type: smart-charge, Delta: -3.61**
  Time: 22:30 - 23:00, Type: smart-charge, Delta: -0.54
  Time: 23:00 - 05:30, Type: smart-charge, Delta: -46.96
[21:43:33.555--INF] Applying charge action to 15 slots for IOG Smart-Charge

[SolisManager-20250323.log](https://github.com/user-attachments/files/19436503/SolisManager-20250323.log)
[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/19436502/SolisManagerExecutionHistory.csv)

Webreaper, 2025-03-25:
The smart-charge slots should be recalculated each time and then the plan re-evaluated, but it sounds like something's missing. It's hard for me to debug because I don't have an EV. I'll try and take a look and see if I can figure out what's happening later this week. :)

Webreaper, 2025-03-25:
I think I might know the cause - I've pushed a speculative fix to dev. If you're using docker, try the `dev` tag, or otherwise pick up the build artifacts from here: https://github.com/Webreaper/SolisAgileManager/actions/runs/14055956268

Let me know if that's better!

Webreaper, 2025-03-27:
Any joy in testing this? Has it solved the problem?

cs95dtt, 2025-03-27:
Sorry for the delay.  I've applied the dev build but haven't got round to
doing a smart charge yet, I usually only charge once a week but will be
doing one this weekend.

I'll update you.

Dan.

Webreaper, 2025-03-27:
No prob, just wanted to check in. Let me know how it goes at the weekend and if it's looking better I'll release it.

closed 2025-03-30 by cs95dtt