Webreaper/SolisAgileManager #158: Daylight Saving Bugs Go here

Webreaper, 2025-03-28:
Creating an issue here in case there's any bugs in my DST changover code.... :)

With luck I'll just close it but I might have to do a release sharpish if peoples' charging happens at the wrong time!!

tabannis, 2025-03-29:
[image]

[image]

Oddness with slots?

Webreaper, 2025-03-29:
What's odd about it?

tabannis, 2025-03-29:
I think the way the octopus app shows the time change (the 12.30-2:00 slot) Vs how the app shows it confused me.(That and it's Friday night 🤪)

If it all looks right, Ignore me, the application is awesome, and I'm just being a "user" 🥴

Webreaper, 2025-03-29:
Oh, I see, yes.

Interesting. Might be that Octopus are providing UTC times from their API, and I need to convert. Will have to see if it corrects itself. Let me check.

Webreaper, 2025-03-29:
Okay, pushed a fix - let's try once it builds. 

[image]

Webreaper, 2025-03-29:
First fix released. Let me know if there's any more!!

By-tor, 2025-03-30:
it's 02:28 and my app (1.0.562) is showing 01:00-01:30

[image]

Webreaper, 2025-03-30:
Make sure you refresh the UI too.

I need to adjust the date summary line so it shows local, not UTC, times too.

tabannis, 2025-03-30:
All looks fine and dandy on v1.0.562.0 after a browser refresh, Thanks!

Webreaper, 2025-03-30:
Another issue for me: the inverter was saying 05:30 at 06:30BST. Seems the inverter time update (which is scheduled for 2am) didn't trigger because of the time change. So I restarted, and now thanks to the inverter time update at startup the inverter time is correct. But it looks like the charging slots are being sent wrong now. Have committed a fix and testing....

Webreaper, 2025-03-30:
So... if I enable real-time updates (the new solis feature) it fails, I think because the inverter is in the wrong time-zone. This is confusing.

dqj999, 2025-03-30:
I downloaded build 562 just and restarted. I noticed that the quoted rate for Intelligent Go was wrong for 3:30 to 5:30 in the graph. 

[image]

Webreaper, 2025-03-30:
Yes, it doesn't surprise me that the rate was wrong if you didn't update to the new version before the switchover. 

Can I ask everyone to try something:
1. Check the inverter state is in 'Do Nothing'
2. Do a manual override to discharge for the 'current' slot
3. See if the inverter starts discharging
4. Then revert to 'Do Nothing'. 

I did this and nothing happened, because the charge instruction was set for a time in the past (it set the discharge for an hour ago).

Once you've done that, please pull the latest dev version and try again. I'm running that and charge insrtuctions are being applied correctly, it seems. 

If that works as expected, I'll do another release.

dqj999, 2025-03-30:
I just did that with build 562 but I set it to charge rather than discharge. The inverter started charging the battery right away.

Webreaper, 2025-03-30:
I have no idea what's going on then. 🤣

Can you try 566 and see if it still works? (it works for me).

dqj999, 2025-03-30:
Yes sorry - meant to use the dev build but accidentally used the last production build!

Just reran the charge/discharge test with build 566 and the inverter responded immediately.

Webreaper, 2025-03-30:
Thanks.

I guess if 566 and 562 both work, I'll hold off doing another release at the moment.

I need to analyse the Solcast data and make sure it's using BST too. But the garden is calling so that might need to wait until tomorrow! 😁

tabannis, 2025-03-30:
Can confirm everything is behaving as expected on my Inverter also.

Webreaper, 2025-03-30:
Phew. Thanks all!

Webreaper, 2025-03-30:
Need an IOG slot to make sure they're working right too. Anyone fancy taking a long drive this morning and charging this afternoon? 😁

dqj999, 2025-03-30:
Will do later.

cs95dtt, 2025-03-30:
Interested to see if @dqj999 or anyone else experiencing the same thing where IOG dispatches slots not being detected.   Not sure if it's just me or a wider issue with Octopus API not updating.

I noticed in my case I plugged car in around 6pm and received a slot for 22:30-23:00 on Ohme app but that wasn't picked up at all for over 2 hours and then later the the first slot changed to 21:30-21:45 according to Ohme but still nothing detected when Solis Manager polls Octopus API.

Not sure if Octopus is having issues its been quite reliable up until last night and today.

dqj999, 2025-03-30:
> Interested to see if [@dqj999](https://github.com/dqj999) or anyone else experiencing the same thing where IOG dispatches slots not being detected. 

You need to check with your car charger when it actually charged the car. The schedule could have changed and therefore there might have been no lower price power at the times of the original schedule. My recent experience has been that when the car was actually being charged that Solis Mannager knew about it and charged the house battery.

cs95dtt, 2025-03-31:
I did check my car when it started to charge at the IOG slot but Solis
manager didn't pick this up, I left it for a good 30mins to see if any of
the subsequent IOG refreshes found these but sadly not.

I switched back to the version of Solis Manager prior to DST fix and that
too didn't pick up the IOG slots.

I don't think the issue is with Solis Manager and more to do with Octopus
dispatches not syncing properly.   I did notice Ohme app was miss reporting
the finish charge time last night by a hour out but this morning it appears
to have been fixed so I'm guessing possible glitch on Ohme/Octopus end.

I'll retry EV charging again and see what happens.  I have a ticket with
Octopus API team open.

Webreaper, 2025-03-31:
Thanks. Think I need to convert the times to local in the charts. Will fix

By-tor, 2025-04-08:
Hi Mark,

My two virtual machines are running on BST but are setting their associated inverters an hour behind causing all the charge slots etc. to be an hour out. Is this an "undocumented feature" or a bug?   :-)

 
[image]

Webreaper, 2025-04-08:
Are you sure the OS timezone is correct on the VM? I don't see this issue running on my NAS, and nobody else has reported it.

tabannis, 2025-04-08:
Ah, so it's not just me. I'm having the same issue as the @By-tor by the looks of it.

[image]

I set my inverter time using the "follow phone time" option over the weekend and now I'm back to being an hour behind again. 

I'll switch to my laptop and see if I can find some logs that show this...

Webreaper, 2025-04-08:
I don't think you want "follow phone time". If you do, you probably need to turn off the setting to set the time in SolisManager.

Have you turned on Daylight Savings in the app?

tabannis, 2025-04-08:
Yup...

[image]

Follow phone time just synchronises the inverter with the phone clock as a one off 👍

Webreaper, 2025-04-08:
No idea then. It's all working here. 😁

tabannis, 2025-04-08:
Ah, the docker instance is also delayed..

tabannis, 2025-04-08:
Fixed it! Added this to my docker-compose.yaml
```
environment:
            - TZ=Europe/London
```

[image]

Webreaper, 2025-04-08:
Ah, nice. I have that in mine, I think. That's why it works for me. I'll
update the docs.

By-tor, 2025-04-08:
Ah, thanks @tabannis  I didn't know the docker container had it's own time zone - I assumed it inherited it from it's host 

Every day's a school day - even at 68 :-)

Webreaper, 2025-04-08:
> And I've been working on computers since 1978 !!!

About 5 years longer than me then. 😆

tabannis, 2025-04-08:
> Ah, thanks @tabannis  I didn't know the docker container had it's own time zone - I assumed it inherited it from it's host 
> 
> Every day's a school day - even at 68 :-)
> 
> And I've been working on computers since 1978 !!!

You're welcome, I only clocked it because some of my other containers have the TZ environment variable defined. 

I'm a relative whipper-snapper then at 41 😆

Webreaper, 2025-04-27:
Going to optimistically close this now, as I think all of the DST bugs are solved now!

closed 2025-04-27 by Webreaper