Webreaper/SolisAgileManager #159: Refused to connect

By-tor, 2025-03-29:
I am trying to set up the app on Debian 11 running docker in a virtual machine.
I have got everything running, as far as I can tell.

[image]

It works fine when I ssh or scp in to it but when I try to connect to the app using port 5169 it says it refused to connect.

I am pretty sure it's something I haven't got set correctly but can't find what !!
Any pointers as to what to check?

Webreaper, 2025-03-29:
What's your docker command or Docker compost?

By-tor, 2025-03-29:
Docker compose

Webreaper, 2025-03-29:
No, I mean can you copy it here so I can see what you've specified in the Docker compose.

By-tor, 2025-03-29:
services:  
    solismanager:
        container_name: solismanager
        image: webreaper/solisagilemanager:latest
        ports:
            - '5169:5169'
        restart: unless-stopped
        volumes:
            - /volume1/dockerdata/solismanager:/appdata

Webreaper, 2025-03-29:
Looks fine to me. Is there an issue with port 5169 on the VM?

By-tor, 2025-03-29:
I don't think so but not sure how I can check.

I have tried nmap to see which ports are open but not clear what the results are telling me

[image]

Webreaper, 2025-03-29:
You're past my Unix knowledge at this point. Probably a question for reddit or stackoverflow.com.

By-tor, 2025-03-29:
Ok - thanks anyway

closed 2025-03-29 by By-tor