Webreaper/SolisAgileManager #160: Incorrect data from inverter

dqj999, 2025-03-31:
These incorrect values were received from my inverter last night. Several values were negative. I don’t think this is a bug in Solis Manager but the negative values messes up the axis values on the history graph. I’ll have to edit the .csv file to fix it. Just wondered if anyone else has seen this.

[image]

dqj999, 2025-03-31:
I don’t normally use scheduled discharge and another unusual factor may be a sufficiently large solar forecast to affect the charging actions.

dqj999, 2025-03-31:
None of the 4 affected columns should be negative. I don’t think that the source of the negative values is the data, I think some code is actually inserting those values. It doesn’t seem to be the case of the wrong columns being updated though as I can’t see which columns that data would apply to.

dqj999, 2025-04-01:
There is a definite problem here. The situation repeated itself last night after a scheduled discharge period ended.

[image]

dqj999, 2025-04-01:
Here’s the schedule 

 
[image]

Webreaper, 2025-04-01:
I suspect it's something relating to the discharge. The inverter numbers are DC and show -ve for discharge, and +ve for charge. I probably need to take account of that more carefully in the logic the processes them. 

There's no adverse side-effects from this, in terms of actual behaviour, right? It's just that it looks odd/wrong in the graph and history view?

dqj999, 2025-04-01:
Yes - no operational impact. It’s just the negative Solcast  data that messes up the choice of the vertical axis for the Solar forecast and yield.

dqj999, 2025-04-03:
[image]

It may not be connected to the discharge times as the same thing happened at 23:00 when I moved the discharge times away from that. Could this be a hangover from BST changes?

Webreaper, 2025-04-08:
I think I might know what is going on here. @harryflatter raised an issue where enrichment is being done because a negative value is cancelling out the day's total. I'll investigate when I have time.

dqj999, 2025-04-11:
Yesterday I removed the login details for Solcast so that no forecasts were created, however I still got a bunch of negative values in the history log at 23:00. 

Other people are reporting the same error. Are they all using the Intelligent Go tariff?

HarryFlatter, 2025-04-11:
I'm on GO, not (very) Intelligent :-)

dqj999, 2025-04-11:
I'm using a large number of planned actions (14). Is that a possible common factor with people experiencing these errors?

Webreaper, 2025-04-11:
Can somebody who's seeing the issue please attach the executionHistory.csv to this issue, so I can see the data that's causing the graph to go wonky?

dqj999, 2025-04-11:
This was the line in my csv file from last night that had the error:
10-Apr-2025 23:00, 10-Apr-2025 23:30, 28.96, DoNothing, MostExpensive, 45%, -15.28, 0.00, -7.44, -10.48, -12.24, 38.05, "Peak price slot - avoid charging"

This is the only error in the csv file because I've zeroed out the other incorrect entries. I've attached the entire CSV file 

[SolisManagerExecutionHistory.csv](https://github.com/user-attachments/files/19702671/SolisManagerExecutionHistory.csv)

Webreaper, 2025-04-11:
AI generated code? 

*points at door*

OUT!!!

😆

HarryFlatter, 2025-04-11:
🤣

By-tor, 2025-04-13:
Did the negative values cause my batteries to be charged last night?
The history page shows forecast PV of 4.78 kWh but the forecast PV on the charging plan for today shows 12.35 kWh.
Charging threshold for overnight charging is set at 9 kWh.

Webreaper, 2025-04-13:
Negative values are a reporting issue, not part of the strategy.

I suspect the forecast changed overnight. The logs should show you if it updates.

dqj999, 2025-04-18:
I think that the error data gets inserted at mid-night. It would be nice to see this bug finally fixed!

Webreaper, 2025-04-18:
Yeah, I'll investigate it soon - the last two weeks have been crazy with
work and personal stuff. Hope to get it fixed soon!

tabannis, 2025-04-24:
A side comment about this, I've been monitoring the Solis API responses recently as part of a (unrelated to this app) home assistant energy dashboard integration and notice there's all sorts of "oddness" in the data going on between 23:00 and 00:00 (BST) each day including values for battery charge and discharge (a change from reporting in integer values to decimal values) but just for that 1 hour 🤷‍♂️

Trying to get an answer out of solis about their API is like trying to herd cats, so who knows what's on.

tabannis, 2025-04-24:
My best guess is some sort of daily "reset" code is running at the Solis end to bring daily cumulative totals back to zero by issuing a negative value, but it's now running at 23:00 rather than midnight.

This is backed up by the negative values given by the API exactly matching the daily cumulative usage for "Actual PV", "Imported kWh" and "Exported kWh".

I notice in history, in the diary table view. The figures for those above values all now read zero...

[image]

Webreaper, 2025-04-24:
Are you going to raise an issue with Solis to report this?

dqj999, 2025-04-29:
Can a temporary fix be implemented to zero the negative values so that the solar forecast graphs have the correct scaling?

Webreaper, 2025-04-29:
It's the next thing I'm going to investigate - it happens to me too...

Webreaper, 2025-04-29:
Added a fix today that a) filters erroneous data from the graph, so it should look better, and b) should filter bad data (negative Actual Yield values) from being saved in the execution history.

Let me know if it's better - it's improved things for me.

tabannis, 2025-04-30:
An improvement but still some odd negative values...

[image]

[image]

Others (on other projects) are trying to get solis to fix their API but aren't getting very far.

Webreaper, 2025-04-30:
One thing we've noticed - some of the figures are reading 'short'. E.g., my history is showing the total Actual yield as 39kWh yesterday, when we actually did 49kWh. 

I'll investigate this when I get a chance. I think it may be rounding errors (long story), but unsure at present.

dqj999, 2025-04-30:
> Okay, put a fix in just now to:

> Not return bad values from the inverter

This seems to have fixed the display issue.

The bad data was for the SolCast forecast as well as the inverter, which indicates that some kind of corruption was going on, possibly triggered by an unrelated event?

Webreaper, 2025-04-30:
I think you're getting mixed up with the bad data caused by the changes related to the code refactor for the consumption/costs functionality. That caused bad records in the history (not just negative values, but incorrect times etc) and was fixed last night.