Webreaper/SolisAgileManager #163: Issue with schedule and home screen.

itolson, 2025-04-03:
So i set the schedule to discharge from 18:30 - 22:30

[image]

but the home screen shows 22:00 - 22:30 as do nothing 

[image]

just wondering if I am doing something wrong ?

Thanks

Webreaper, 2025-04-03:
From that it looks like a bug. I'm on holiday at the moment so won't be able to investigate until Monday....

Webreaper, 2025-04-03:
Can you let me know what happens when it gets to 2230? It may be that the next refresh updates correctly (in which case it's a less critical display issue - since nothing actually happens until the slot is "now").

itolson, 2025-04-03:
I will yes but I just noticed that the home screen shows 22:30 - 23:00 and 23:00 to 23:30 which i have not set

itolson, 2025-04-03:
It didn't discharge at 22:00

closed 2025-04-27 by Webreaper