Webreaper/SolisAgileManager #165: Intelligent Octopus Go EEPROM writes

abrasive-blueberry, 2025-04-06:
I read the bit about reducing EEPROM writes and have a suggestion. It won’t apply to everyone.

I am on the IOG tariff and use the app to monitor for extra charge slots (outside the 23:30 to 05:30 window), and set the inverter to charge during these slots. All working great.

However I don’t charge the car every night, maybe twice a week, so for the rest of the time the inverter only needs to be set to charge 23:30 to 05:30.

The behaviour at the moment is that the app writes a 23.30 to 05:30 charge every night and deletes it again in the morning.

However, could it be implemented so that one of the inverter’s 3 programmable charge slots (eg slot 2) is permanently used for the fixed charging periods (23:30 to 05:30) thus saving the need to write and remove it every day, and slot 1 is used for the additional dynamic charging periods?

Webreaper, 2025-04-06:
It could do. And yes, with some complex algo I could use the multiple slots to reduce writes even further. But the complexity would be huge, and would likely result in conflicts sometimes, which would end up with things failing. So I elected to just keep it to using one slot, as the logic is simpler and will be more reliable and easier to debug.

The reduction in EEPROM writes wouldn't be that significant - it would save a few a week. My main goal was to not do what most of the inverter control scripts do, which is to write the new state every 30 minutes - which is unnecessarily heavy on the inverter writes.

abrasive-blueberry, 2025-04-06:
Fair enough, it’s possibly me just overthinking things. Your app is brilliant and this wasn’t a criticism :)

It’s only 2 writes a day, so as you said, it wouldn’t save anything significant and probably isn’t worth the complexity involved.

Webreaper, 2025-04-06:
No, don't worry, I wasn't taking it as a criticism. I have thought a lot about how I could set multiple slots to make everything more efficient, but figuring out the conflicts etc just made my head hurt.

So you're 100% correct - it's just a non-trivial problem to solve.

Maybe one day in future!

closed 2025-04-24 by Webreaper