Webreaper/SolisAgileManager #167: Graph to show net import costs using Octopus API

Webreaper, 2025-04-10:
Octopus have separate graphs for import and export. All I care about is did I pay for electricity on any given day, and if so how much. If not, how much profit did I make.

Webreaper, 2025-04-29:
First cut is available in dev. 

[image]

Still to do: 
- Chart
- Date Ranges
- Option to add standing charge into the calculation
- Some better caching around tariff data

Webreaper, 2025-04-29:
[image]

dqj999, 2025-04-29:
1. The headings for the Export (kWh) and Import costs need to be swapped.
2. No allowance is made for the price of Smart Daytime charging being 7p instead of full rate on IOG. 

   (This is not showing in the History either - although I thought it was at one point).

Webreaper, 2025-04-29:
Thanks.

Octopus's API doesn't return any info about IOG slots and reduced prices, unfortunately. The numbers showing in this new table/chart exactly match what's in the Octopus estimated costs graphs/page. 

Does that page show reduced costs due to IOG for you? I suspect not - I think the IOG discounts are only applied when the bill is calculated.

Webreaper, 2025-04-29:
Oh, and I didn't realise you can zoom with the mouse wheel. That's a feature of the charting component I use!

dqj999, 2025-04-29:
> Oh, and I didn't realise you can zoom with the mouse wheel. That's a feature of the charting component I use!

It doesn't work on the History pages for some reason - would be nice if it did.

I'm getting a few issues with this dev build - I think it is in GMT mode. The history page is showing up to an hour in the future!

I'm going to revert to the production version.

[image]

Webreaper, 2025-04-29:
I had to refactor a lot of code to do the costs screen, so may have messed up the DST adjustments. Will fix those in the next couple of days.

dqj999, 2025-04-29:
I've reverted to the Production version but I'm still getting errors in the History, both the graph and the table. 

I'm wondering if the dev build has messed with some stored data.

[image]

[image]

Webreaper, 2025-04-29:
Yep, could have done....

tabannis, 2025-04-29:
I've spun up a container for dev, cloned the config files from my "latest" environment, then set simulation mode in dev.
The new costs area gives an error...

[image]

...Can't see any errors in the .log files. Tried renaming the SolisManagerExecutionHistory.csv to SolisManagerExecutionHistory.old to see if that helps, restarted the container,costs screen just now sticks at "Loading 30 days of consumption and tariff data..." and logs now have errors (attached) and SolisManagerExecutionHistory.csv hasn't been regenerated.

[SolisManager-20250429.log](https://github.com/user-attachments/files/19968110/SolisManager-20250429.log)

Did i make dev angry by cloning my config from latest?

Webreaper, 2025-04-29:
Just been ironing out some bugs with the restructured code for getting tariff rates. I think it's fixed now, if you pull latest it should work. (literally fixed these in the last 15 mins).

Webreaper, 2025-04-29:
BTW, in case you're interested this bug happens because the new code loads the full set of tariff data which goes back to Feb. However, the octopus API returns two results for the same time on the point of the DST switchover - so you get 01:00 and 01:30 twice..... Confused me for a while!

I think this should all be good and working again now - please test the latest dev build and let me know if you see anything else odd.

tabannis, 2025-04-30:
Yep, that's sorted it all out 👍 Thanks!

Webreaper, 2025-04-30:
Closing this - will raise some related issues for feature enhancements.

closed 2025-04-30 by Webreaper