Webreaper/SolisAgileManager #169: [enhancement] IoGo charge rate option

exalyte, 2025-04-17:
Firstly thanks for the brilliant tool! 

Is it possible to add an override to the IoGo settings to say when charging outside of the normal 2330-0530 window to use a different charging rate? 

I currently charge 2330-0530 at 53amps as this stops the inverter fans kicking in (its in the garage and the frequency of the fans is ohh just so that it can be heard in the upstairs bedroom) however tonight for example octopus decided to charge my car for 30 mins at 2100 the house battery did charge as expected but only at the 53amps, at this time i wouldn't mind it going at 100amps as is its limit. 

Example;
If IoGo override (outside 2330-0530) charge @ xxx amps, just to allow full utilisation of the charging windows, more of an issue in winter with the longer days. 

[image]

Thanks again and appreciate the time to review

Cheers,

Exa

Webreaper, 2025-04-18:
Nice mockup. 😁

Yeah, should be easy.

exalyte, 2025-04-18:
Sweet thank you

Webreaper, 2025-04-18:
Okay, that should be there in the latest dev build: https://github.com/Webreaper/SolisAgileManager/actions/runs/14542218939 (if you're using docker, pull `webreaper/solisagilemanager:dev`). Can you test and see if it does what you need?

exalyte, 2025-04-18:
Sweet thanks

ill give it a few days (cars fully charged now) and let you know 

but it looks better than my mockup :p 

[image]

Thanks reaper!

exalyte, 2025-04-18:
quick update, usual overnight charge just started at the expected 53amps 👍 
Will feed back when i get an out of hours charge to test. 

thanks again 
Exa

exalyte, 2025-04-19:
Utterly unrelated but as we've been chatting 
What happens on iogo if you have a power cut, will it still change it to charge or is it clever enough to realise it's in no grid mode, I ask because I have a powercut right now and I've forced it to no action untill 4am which is when the dno say it'll be back on but just wondering if it'll need micromanaged or not...? 

Thanks again

Webreaper, 2025-04-20:
The app doesn't do anything specific for a no-grid situation.

Can you elaborate more on what you'd like it to do if it was possible?

exalyte, 2025-04-20:
I assumed that if the inverter was set to charge the batteries during a nogrid situation it would stop providing power to the backup side, as the batteries are either in or out not both. 

I'm not sure what would happen (didn't fancy trying it as both options required a good 20 mins of clean shutting down and then rebooting and I was moderately piddeled by the time I got home to no power as was 😂) so I can't say for certain if the inverter itself would opt out.

I guess my ask is during a nogrid situation can the app "do nothing" during the usual charge windows to avoid disruption on the backup side.

Webreaper, 2025-04-20:
I think we'd have to check (or ask) to be sure. I also use AC backup for the router/modem/NAS/etc and we have frequent power cuts - but I've not tried charging during a power cut so no idea. It feels like AC backup would be prioritised - that would be sensible. My understanding is that if there's no grid:

- The AC backup will be powered from PV if there is any
- The AC backup will use what's in the battery next

But whether the batteries can be charged while there's no grid and the AC backup is being powered, I'm not sure. I feel like it could be though - for an off grid inverter the batteries can power the inverter while it's charging, I thought? But unsure.

Webreaper, 2025-04-20:
Actually, it must be fine.

1. If there's no grid, and there's power in the battery it'll be used for the AC backup

2. If there's no grid and there's the same PV generation as the AC Backup load, the backup will be covered directly from PV without draining the battery.

3. If there's more PV than the AC Backup load, then the backup will be covered by PV and any excess will go into the battery.

I don't think it'll make any difference whether the inverter is explicitly set to charge, because the inverter **always** covers the house load (including AC Backup) first, before charging or exporting, if you're in self-use mode.

dqj999, 2025-04-20:
when I had an unexpected power cut, I found that the History Graphs had some odd values recorded when the system restarted suggesting that the data initialisation had a few issues. Luckily this is easier to test than disconnecting from the grid! 😊

exalyte, 2025-04-23:
Hello!

Sorry Easter weekend and then I got unwell and had a short stint in the old hospifail 

Regarding power cuts I can't comment I manually set it to not charge as to avoid any unforeseen outcomes
Perhaps I should test this manually one day and see what the verdict is. 

With the IOGo enhancement you have made it's been fine for a few days, car was in a carpark so not at home to charge, but I've plugged the car in this evening and it started charging at 53amps as usual however at midnight the IOGo charge window opened and the inverter ramped to 100amps. 
Is it possible to add a logic gate here if time is <2330 default to IOGo override if >2330 standard charge current applies

Sorry i have no clue how easy/hard this is to do im no programmer 🤣 

Thanks again for the on going support! 

Cheers
Exa

closed 2025-04-27 by Webreaper